chrisss404/check-mk-arm #4: Missing armv7l_linux.mk

makanakoeln, 2018-06-06:
I get:

make[3]: armv7l_linux.mk: Datei oder Verzeichnis nicht gefunden
make[3]: *** Keine Regel, um �armv7l_linux.mk� zu erstellen.  Schluss.     make[3]: Verzeichnis �/home/pi/make_check_mk/check-mk-raw-1.4.0p21.cre/packages/snap7/snap7-full-1.3.0/build/unix� wird verlassen
Makefile:23: die Regel f�r Ziel �build� scheiterte
make[2]: *** [build] Fehler 2
make[2]: Verzeichnis �/home/pi/make_check_mk/check-mk-raw-1.4.0p21.cre/packages/snap7� wird verlassen
Makefile:83: die Regel f�r Ziel �build� scheiterte                         make[1]: *** [build] Fehler 2                                              make[1]: Verzeichnis �/home/pi/make_check_mk/check-mk-raw-1.4.0p21.cre� wird verlassen
debian/rules:24: die Regel f�r Ziel �build-stamp� scheiterte
make: *** [build-stamp] Fehler 2
dpkg-buildpackage: Fehler: Fehler-Exitstatus von debian/rules build war 2

Any idea?

Thx!

makanakoeln, 2018-06-06:
Think i found a solution. Did i miss something or is that patch missing?

flix87, 2018-06-07:
Look here:
https://github.com/chrisss404/check-mk-arm/issues/2

makanakoeln, 2018-06-07:
Works for me, thanks!

closed 2018-06-07 by makanakoeln