chrisss404/check-mk-arm #6: No services in host view using check-mk-raw-1.5.0p4_0

onexp, 2018-09-26:
Hi,

I just installed your latest version check-mk-raw-1.5.0p4_0.stretch_armhf.deb on my Raspberry pi 2. As OS I am using Raspbian Strech Lite.

So far the installation went fine, sites are running and I also installed the agent on some test servers running Ubuntu 16.04. All services were detected with auto discovery and added. 

But something is wrong with the service views. If I go to the "All host" section I can see my servers and the overview, how many services are OK, WARN, etc.  But if I click on a host, the service view is emtpy. Only for the localhost the view is working. There I can see all services.  
But if I go to Main Overview and click under "Service Statistics" on OK, or Total, I get a complete overview of my services, sort by host (same if I go to "All services". 
I was not able to find the problem and installed your version check-mk-raw-1.4.0p35_0.stretch_armhf.deb This version is running fine and all views are working as expected.

chrisss404, 2018-09-26:
Hi,

I currently monitor 4 hosts and the service view works as expected for all of them.

Things you can try:

* Update host services: WATO > Hosts > Edit the services of host > Automatic refresh (tabula rasa)
* A fresh install

### My setup

    $ lsb_release -a
    No LSB modules are available.
    Distributor ID:	Raspbian
    Description:	Raspbian GNU/Linux 9.4 (stretch)
    Release:	9.4
    Codename:	stretch

    $ omd version
    OMD - Open Monitoring Distribution Version 1.5.0p4.cre

onexp, 2018-09-27:
Thanks for your feedback. 
Update and fresh install didn't help. I even took another Pi 2 and installed everything from scratch. But I am still not able to see the services, when I go to "All hosts" and click on one host. Then I installed mk_check on another x86 VM Ubuntu 16.04 and I was able to see the services with the new version. So I really don't know, where is the problem. Should be no difference between Pi 3 (you are running) and Pi 2 ?!

gituser-rk, 2018-10-05:
Hello,
same strange behavior here (Raspi 3B). The view of one host shows no services, "view.py?view_name=allservices" is showing the services of this host. I have deleted this host and recreated it. The services are shown during the service scan when creating the host in WATO. But later the same - no services in host service view.
In my case this host is a "Fritzbox" router. The services of all other hosts show up fine in the individual host service view. As long as I remember this is since version 1.5.0p2 up to 1.5.0p4. Version 1.4.3x was fine.

chrisss404, 2018-10-07:
Hi,

one last try. The only difference to the upstream build is that python is built without `--enable-shared` flag, as can be seen in python-Makefile.patch. There are suggestions to do this in order to improve performance. However, if this introduces issues it should be removed.

[Here](https://www.dropbox.com/s/oy2p22is3qng266/check-mk-raw-1.5.0p5_0.stretch_armhf.deb?dl=0) is a build without applying the mentioned patch.
Hopefully this will fix the issue.

gituser-rk, 2018-10-08:
Hello,

tried the "special" build and it made no difference. Just installed the .deb file over and started the OMD instance.

mwiedemann, 2018-10-09:
I have exactly the same problem. RPi3 with a fresh install. If I have created a host an added some services the service view is empty:
Linkname: http://cmk/default/check_mk/view.py?host=myHost&site=default&view_name=host#

chrisss404, 2018-10-10:
Some more input that hopefully helps narrow down the issue.

### Logs
Have you checked the logs of your site? Maybe there is a hint what is causing the issue.

    /opt/omd/sites/<your-site>/var/log/apache/error_log
    /opt/omd/sites/<your-site>/var/log/apache/access_log
    /opt/omd/sites/<your-site>/var/log/web.log

### Hostnames
Maybe the issue is related to the used hostname. Here is an overview of my hosts / hostnames:

| Hostname  | Url | Services shown |
| --------- | --- | -------------- |
| htz1 | /check_mk/view.py?host=htz1&site=health&view_name=host | Yes |
| rpi | /check_mk/view.py?host=rpi&site=health&view_name=host | Yes |
| up | /check_mk/view.py?host=up&site=health&view_name=host | Yes |
| wt | /check_mk/view.py?host=wt&site=health&view_name=host | Yes |

### Dependencies
Maybe I have installed a package, which is not defined as dependency in the .deb package.

[Here](https://github.com/chrisss404/check-mk-arm/files/2465909/packages.txt) is a list of installed packages of my raspberry pi:

    dpkg --get-selections | grep -v deinstall > packages.txt

msperl, 2018-10-27:
just renaming the hosts changes the behaviour!
* short name (working) -> name with dots: stops to work
* name with dots (not working) -> name without dots: starts to work!
(I have tested wit 1.5.0p2, p4, p5 and p7 - all show the same behavior)

Maybe a python library thingy?

msperl, 2018-10-27:
So I have started tracing it and it seems to be related to the response of live-status:
```
OMD[cmk]:~$ cat /tmp/test.sh
#! /bin/sh

for x in kvm kvm.intern; do
    echo "Processing: $x"
    echo "================="
    echo -n "GET hosts\nColumns: host_custom_variable_names host_action_url_expanded host_filename host_check_command host_active_checks_enabled host_perf_data host_pnpgraph_present host_custom_variable_values host_name host_check_type host_address host_notes_url_expanded\nFilter: host_name = $x\nLocaltime: 1540640697\nOutputFormat: python\nKeepAlive: on\nResponseHeader: fixed16\nColumnHeaders: off\n\n" \
	| nc.openbsd -w 1 -U /omd/sites/cmk/tmp/run/live
done

    echo "list hosts"
    echo "================="
    echo -n "GET hosts\nColumns: host_name host_check_type host_address\nLocaltime: 1540640697\nOutputFormat: python\nKeepAlive: on\nResponseHeader: fixed16\nColumnHeaders: off\n\n" \
| nc.openbsd -w 1 -U /omd/sites/cmk/tmp/run/live
OMD[cmk]:~$ sh /tmp/test.sh
Processing: kvm
=================
200         506
[[[u"ADDRESS_4",u"ADDRESS_6",u"TAGS",u"FILENAME",u"ADDRESS_FAMILY"],u"/cmk/pnp4nagios/index.php/graph?host=kvm&srv=_HOST_' class='tips' rel='/cmk/pnp4nagios/index.php/popup?host=kvm&srv=_HOST_",u"/wato/hosts.mk",u"check-mk-host-ping!-w 200.00,80.00% -c 500.00,100.00%",1,u"rta=0.586ms;200.000;500.000;0; pl=0%;80;100;; rtmax=0.629ms;;;; rtmin=0.541ms;;;;",1,[u"10.10.10.250",u"",u"/wato/ cmk-agent ip-v4 ip-v4-only lan no-snmp prod site:cmk tcp wato",u"/wato/hosts.mk",u"4"],u"kvm",0,u"10.10.10.250",u""]]
Processing: kvm.intern
=================
200           3
[]
list hosts
=================
200          64
[[u"kvm",0,u"10.10.10.250"],
[u"kvm.intern",0,u"10.10.10.250"]]
```

So it seems to be related to live-status!

On the intel instance I get:
```
Processing: kvm
=================
200         506
[[[u"ADDRESS_4",u"ADDRESS_6",u"TAGS",u"FILENAME",u"ADDRESS_FAMILY"],u"/cmk/pnp4nagios/index.php/graph?host=kvm&srv=_HOST_' class='tips' rel='/cmk/pnp4nagios/index.php/popup?host=kvm&srv=_HOST_",u"/wato/hosts.mk",u"check-mk-host-ping!-w 200.00,80.00% -c 500.00,100.00%",1,u"rta=0.316ms;200.000;500.000;0; pl=0%;80;100;; rtmax=0.495ms;;;; rtmin=0.167ms;;;;",1,[u"10.10.10.250",u"",u"/wato/ cmk-agent ip-v4 ip-v4-only lan no-snmp prod site:cmk tcp wato",u"/wato/hosts.mk",u"4"],u"kvm",0,u"10.10.10.250",u""]]
Processing: kvm.intern
=================
200         527
[[[u"ADDRESS_4",u"ADDRESS_6",u"TAGS",u"FILENAME",u"ADDRESS_FAMILY"],u"/cmk/pnp4nagios/index.php/graph?host=kvm.intern&srv=_HOST_' class='tips' rel='/cmk/pnp4nagios/index.php/popup?host=kvm.intern&srv=_HOST_",u"/wato/hosts.mk",u"check-mk-host-ping!-w 200.00,80.00% -c 500.00,100.00%",1,u"rta=0.718ms;200.000;500.000;0; pl=0%;80;100;; rtmax=0.769ms;;;; rtmin=0.648ms;;;;",1,[u"10.10.10.250",u"",u"/wato/ cmk-agent ip-v4 ip-v4-only lan no-snmp prod site:cmk tcp wato",u"/wato/hosts.mk",u"4"],u"kvm.intern",0,u"10.10.10.250",u""]]
list hosts
=================
200          64
[[u"kvm",0,u"10.10.10.250"],
[u"kvm.intern",0,u"10.10.10.250"]]
```
So here livestatus reports information for both versions...

msperl, 2018-10-28:
So my trial was successfull - compiling with 7.3 and the "." problem is gone...

msperl, 2018-10-28:
For those of you wanting to give it a try to see if it solves the issue here my deb package:
https://github.com/msperl/check-mk-arm/releases/tag/1.5.0p7-gcc7.3-test

chrisss404, 2018-10-28:
@msperl thanks for your effort. 

Can you please share your steps how to resolve this issue. I don't know what is meant by a "slow" compile of gcc 7.3.0, can you explain it? Did you compile gcc 7.3.0 from source or did you use a pre-compiled binary, if so which one?

msperl, 2018-10-28:
I have just compiled gcc the way our original source gcc was compiled (just replaced the version for all practical purposes)

I am actually compiling the whole as a series of docker containers - this should become available in my clone of the repo soon.

msperl, 2018-10-29:
here the initial version that builds check_mk via a series of docker images
https://github.com/msperl/check-mk-arm/tree/dockerize

I am thinking that maybe the default docker images of GCC would also be valid so that we do not have to compile gcc on our own... (but that may leave out the rpi 1 and 2's - to get tested...)

I am looking into that as soon as I have finished building a docker image for running cmk based on the Dockerfile from check_mk itself...

msperl, 2018-10-29:
the above branch now also contains the dockerized version of check_mk using the same basic image Dockerfile  that checkmk provides with its Sourcecode (besides the removed gpg verification of the debian package, which the build processs currently does not support...).

So you can run:
```
docker container run -dit -p 8080:5000 -v /omd/sites --name check_mk -v /etc/localtime:/etc/localtime --restart always msperl/check_mk:1.5.0p7-7.3.0-arm
```

to test the image inside a docker environment on arm - more information on the image can be found here: https://hub.docker.com/r/checkmk/check-mk-raw/ for details...

chrisss404, 2018-11-30:
Should be fixed with 1.5.0p8.

closed 2018-11-30 by chrisss404