chrisss404/check-mk-arm #10: Raspberry Pi 4 Update

su-overlord, 2019-07-13:
Hi could this be updated for Raspberry Pi 4?

chrisss404, 2019-07-18:
Hi there,

is it not working for the Pi 4? 

If not, they might use the latest version (buster) of Debian. Currently the check_mk deb packages are build for the previous version (stretch). I have two solutions for you, either you build check_mk yourself by running the build script, or you wait until I have updated my Pi to the latest version of Debian so that I can build such packages. However this will take some time.

BR
Christian

chrisss404, 2019-08-05:
Good news. The latest release (1.5.0p21) is targeting Debian Buster please check if it is working on your RPi4. 

Please reopen this issues in case you are running into issues.

closed 2019-08-05 by chrisss404