chrisss404/check-mk-arm #11: site creation fails on fresh raspian buster install

chris240189, 2019-08-10:
I did a fresh install of raspian buster and the site creation fails with a python error:

`pi@raspberrypi:~ $ sudo omd create homelab
ERROR:root:code for hash md5 was not found.
Traceback (most recent call last):
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type md5
ERROR:root:code for hash sha1 was not found.
Traceback (most recent call last):
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha1
ERROR:root:code for hash sha224 was not found.
Traceback (most recent call last):
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha224
ERROR:root:code for hash sha256 was not found.
Traceback (most recent call last):
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha256
ERROR:root:code for hash sha384 was not found.
Traceback (most recent call last):
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha384
ERROR:root:code for hash sha512 was not found.
Traceback (most recent call last):
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha512
Traceback (most recent call last):
  File "/usr/bin/omd", line 4486, in <module>
    from md5 import md5  # deprecated with python 2.5
  File "/omd/versions/1.5.0p21.cre/lib/python2.7/md5.py", line 10, in <module>
    from hashlib import md5
ImportError: cannot import name md5`

chrisss404, 2019-08-10:
Thank you for reporting this issue.

It seems like I am using the wrong version of libssl, I guess it should be `libssl-dev` instead of `libssl1.0-dev`.

Please try to install version 1.0.2 of libssl until I have a new package with the updated dependency ready:

    apt-get install libssl1.0.2

chris240189, 2019-08-10:
After installation of libssl1.0.2 site creation was successful!

thanks

wemsras, 2019-08-11:
Some issue here. After installing libssl1.0.2 OMD works fine.

Do you provide for OMD 1.5.21 and newer a version for Debian "stretch"?

Thanks for your work. I'ts fantastic to get OMD for the Raspberry PI.

chrisss404, 2019-08-12:
Updated package 1.5.0p21 to use the right version of libssl.

> Do you provide for OMD 1.5.21 and newer a version for Debian "stretch"?

No, 1.5.0p20 was the last package for Debian Stretch from my side.

closed 2019-08-12 by chrisss404