chrisss404/check-mk-arm #14: No issue just a question

Dasmonk3003, 2019-09-11:
Hi @chrisss404 ,

I think you work with the Check_MK on Raspberry Pi is awesome.
Was wondering what OS are you using on the RPI, also have you benchmarked it?
Meaning amount of services you are able to run on it before it's crashing.

Thanks again for awesome project.

//Dasmonk

chrisss404, 2019-09-11:
> Was wondering what OS are you using on the RPI, also have you benchmarked it?

[Raspbian](https://www.raspbian.org/). No, I have no benchmarks.

> Meaning amount of services you are able to run on it before it's crashing.

I have not experienced a crash until now. However, currently I monitor only 4 hosts and about 100 services on a RPi3, but I had more in the past (up to 10 hosts and a lot more services).

Let me know if you are looking for specific benchmarks. I can share my OMD health performance graph over the last year, see below.

BR
Christian

[image]

chotaire, 2019-09-25:
If you experience any CPU issues, you could just change the services and snmp check intervals from 1 to 2 minutes, which would almost decrease the CPU usage by 50%. I have sites checking only every 4 minutes, but that really depends on the quality of service you will need to provide. If it's just for your home lab, I am personally fine even with 14 minutes (I use aNag to check every 15 minutes, which will also greatly decrease battery usage).

closed 2019-12-14 by chrisss404