chrisss404/check-mk-arm #15: Checkmk 1.6.0 has been released, will you provide packages?

chotaire, 2019-09-25:
Checkmk 1.6.0 has just been released. Any info about an upgrade yet?

Cheers
Chot

chrisss404, 2019-09-26:
Thanks for the hint.

Yes, I plan to provide packages for 1.6.x. I'll let you know when it's ready.

BR
Christian

narongarj, 2019-09-27:
Hi Christian,
I wish to thank you for your excellent work. I'm not developer and not understand about cross-compiling, so my tries to made check_mk work in Raspi was a totally fail.
Whitout your effort I never will be able to monitoring my network without have a server or a VM. That way, using raspberry 3 (rasp 1 not enough to this days necessities) I can see the servers and switchs and discover a lot of problems that was called "IT fault" before and now I know that was hardware problems, database problems and not Infra problems.
Thank you so much and I hope more and more people recognize your work. If came to Rio de Janeiro sometimes I bough you a beer (a good one, not the brazillian popular beers).
Sincerly,
Vagner Fonseca

chrisss404, 2019-10-03:
Just release [1.6.0p2](https://github.com/chrisss404/check-mk-arm/releases/tag/1.6.0p2).

Have a nice evening.

closed 2019-10-03 by chrisss404

chotaire, 2019-10-14:
Hint, will be released tomorrow: https://github.com/tribe29/checkmk/releases/tag/v1.6.0p4