chrisss404/check-mk-arm #16: Raspberry self-check "OMD %sitename% performance" crashes on 1.6.0p5/Buster

donald2612, 2019-11-08:
Just tried the sources on latest Buster release, and added the monitoring host itself, after installing the agent-package.

It crashes on that check 

"OMD %sitename% performance" 

with 

"ValueError (invalid literal for int() with base 10: '')" 

Other checks run fine though. That points to something wrong with the python-version, but I don't really know.

Any hints?

chrisss404, 2019-11-09:
Hi donald,

thank you for reporting this issue.

I could identify the problem and can suggest two ways to solve it.

* either edit the file `/omd/sites/health/share/check_mk/checks/livestatus_status` as follows

        -    if cert_valid_until is not None:
        +    if cert_valid_until is not None and cert_valid_until != "":

* or wait for the next relase

HTH & BR
Christian

donald2612, 2019-11-27:
perfect solution - thanks Chris!

closed 2019-12-14 by chrisss404

chrisss404, 2020-06-13:
Has been fixed upstream: https://checkmk.de/check_mk-werks.php?werk_id=10632