chrisss404/check-mk-arm #19: check-mk-raw-1.6.0p8 package pre-removal script subprocess returned error exit status 1

mxwi, 2020-02-07:
Hi,

if I try to uninstall Check_mk from my device i get an error. Any hints?

```
pi@raspberrypi:~ $ sudo dpkg --remove check-mk-raw-1.6.0p8
(Reading database ... 78801 files and directories currently installed.)
Removing check-mk-raw-1.6.0p8 (0.buster) ...
Site default is still using this version!
dpkg: error processing package check-mk-raw-1.6.0p8 (--remove):
 installed check-mk-raw-1.6.0p8 package pre-removal script subprocess returned error exit status 1
Errors were encountered while processing:
 check-mk-raw-1.6.0p8
```

chrisss404, 2020-02-07:
Hi mxwi,

you are not able uninstall check_mk because one of your sites is still using the installed version, as you can see in the error message:

> Site default is still using this version!

Try to remove the site first (`omd rm default`) and then try again.

BR
Christian

mxwi, 2020-02-19:
worked perfectly!
Thank you!

closed 2020-02-19 by mxwi