chrisss404/check-mk-arm #20: Need Help - S905W Device - Armbian - Debian Bullseye and Buster

khiro1337, 2020-03-28:
if i follow your installation instructions i get the following errors, if in the last command i agree it removes the whole armbian environment after that checkmk doesn't work either, do you have an idea why? i'm desperate - i just want to have pihole and checkmk running on it :( 

The Box:
A95X F1 - Amlogic S905W
    _    __  __ _         ____  ___
   / \  |  \/  | |       / ___|/ _ \__  ____  ____  __
  / _ \ | |\/| | |   ____\___ \ (_) \ \/ /\ \/ /\ \/ /
 / ___ \| |  | | |__|_____|__) \__, |>  <  >  <  >  <
/_/   \_\_|  |_|_____|   |____/  /_//_/\_\/_/\_\/_/\_\

Welcome to Armbian bullseye with Linux 5.5.0-rc6-aml-s9xxx
root@LBOX:~# cat /etc/issue
Armbian 20.02.0-rc1.038 Bullseye \l
root@LBOX:~# uname -r
5.5.0-rc6-aml-s9xxx
root@LBOX:~# uname -a
Linux LBOX 5.5.0-rc6-aml-s9xxx #rc1.038 SMP PREEMPT Thu Feb 6 09:49:22 MSK 2020 aarch64 GNU/Linux
---
root@LBOX:~# curl -LO $(curl -s https://api.github.com/repos/chrisss404/check-mk-arm/releases/latest | grep browser_download_url | cut -d '"' -f 4)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   631  100   631    0     0   1894      0 --:--:-- --:--:-- --:--:--  1894
100 74.5M  100 74.5M    0     0  8450k      0  0:00:09  0:00:09 --:--:--  9.8M
root@LBOX:~# dpkg -i check-mk-raw-*_armhf.deb
Vormals nicht ausgewähltes Paket check-mk-raw-1.6.0p10:armhf wird gewählt.
(Lese Datenbank ... 35363 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von check-mk-raw-1.6.0p10_0.buster_armhf.deb ...
Entpacken von check-mk-raw-1.6.0p10:armhf (0.buster) ...
dpkg: Abhängigkeitsprobleme verhindern Konfiguration von check-mk-raw-1.6.0p10:armhf:
 check-mk-raw-1.6.0p10:armhf hängt ab von time.
 check-mk-raw-1.6.0p10:armhf hängt ab von traceroute.
 check-mk-raw-1.6.0p10:armhf hängt ab von dialog.
 check-mk-raw-1.6.0p10:armhf hängt ab von dnsutils.
 check-mk-raw-1.6.0p10:armhf hängt ab von graphviz.
 check-mk-raw-1.6.0p10:armhf hängt ab von apache2.
 check-mk-raw-1.6.0p10:armhf hängt ab von libdbi1.
 check-mk-raw-1.6.0p10:armhf hängt ab von libevent-2.1-6.
 check-mk-raw-1.6.0p10:armhf hängt ab von libgd3.
 check-mk-raw-1.6.0p10:armhf hängt ab von libltdl7.
 check-mk-raw-1.6.0p10:armhf hängt ab von libpango1.0-0.
 check-mk-raw-1.6.0p10:armhf hängt ab von libperl5.28.
 check-mk-raw-1.6.0p10:armhf hängt ab von libreadline7.
 check-mk-raw-1.6.0p10:armhf hängt ab von libuuid1.
 check-mk-raw-1.6.0p10:armhf hängt ab von binutils.
 check-mk-raw-1.6.0p10:armhf hängt ab von rpm.
 check-mk-raw-1.6.0p10:armhf hängt ab von php.
 check-mk-raw-1.6.0p10:armhf hängt ab von php-cgi.
 check-mk-raw-1.6.0p10:armhf hängt ab von php-cli.
 check-mk-raw-1.6.0p10:armhf hängt ab von php-gd.
 check-mk-raw-1.6.0p10:armhf hängt ab von php-sqlite3.
 check-mk-raw-1.6.0p10:armhf hängt ab von php-pear.
 check-mk-raw-1.6.0p10:armhf hängt ab von smbclient.
 check-mk-raw-1.6.0p10:armhf hängt ab von rpcbind.
 check-mk-raw-1.6.0p10:armhf hängt ab von lcab.
 check-mk-raw-1.6.0p10:armhf hängt ab von xinetd.
 check-mk-raw-1.6.0p10:armhf hängt ab von libfreeradius3.
 check-mk-raw-1.6.0p10:armhf hängt ab von libpcap0.8.
 check-mk-raw-1.6.0p10:armhf hängt ab von libgsf-1-114.
 check-mk-raw-1.6.0p10:armhf hängt ab von libglib2.0-0.
 check-mk-raw-1.6.0p10:armhf hängt ab von poppler-utils.
 check-mk-raw-1.6.0p10:armhf hängt ab von libffi6.
 check-mk-raw-1.6.0p10:armhf hängt ab von libssl1.1.
 check-mk-raw-1.6.0p10:armhf hängt ab von libpq5.
 check-mk-raw-1.6.0p10:armhf hängt ab von libnl-3-200.

dpkg: Fehler beim Bearbeiten des Paketes check-mk-raw-1.6.0p10:armhf (--install):
 Abhängigkeitsprobleme - verbleibt unkonfiguriert
Trigger für systemd (244.3-1) werden verarbeitet ...
Trigger für man-db (2.9.1-1) werden verarbeitet ...
Fehler traten auf beim Bearbeiten von:
 check-mk-raw-1.6.0p10:armhf
root@LBOX:~# apt-get install -f
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Abhängigkeiten werden korrigiert ... Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  apt-file distro-info-data gir1.2-glib-2.0 gir1.2-packagekitglib-1.0 iso-codes libapt-pkg-perl libbind9-161 libdns1104 libdns1109 libexporter-tiny-perl libfstrm0 libgeoip1
  libgirepository-1.0-1 libiptc0 libirs161 libisc1100 libisc1105 libisccc161 libisccfg163 liblist-moreutils-perl liblmdb0 liblwres161 libmaxminddb0 libmpdec2 libpackagekit-glib2-18
  libprocps7 libprotobuf-c1 libpython3-stdlib libpython3.7-minimal libpython3.7-stdlib libpython3.8-minimal libpython3.8-stdlib libregexp-assemble-perl python-apt-common python3.7
  python3.7-minimal
Verwenden Sie »apt autoremove«, um sie zu entfernen.
Die folgenden zusätzlichen Pakete werden installiert:
  apache2-utils libapr1 libaprutil1 libfribidi0
Die folgenden Pakete werden ENTFERNT:
  armbian-config bind9-host check-mk-raw-1.6.0p10:armhf command-not-found dialog dnsutils iotop linux-bullseye-root-current-aml-s9xxx lsb-release networkd-dispatcher python3
  python3-apt python3-dbus python3-distro-info python3-gi python3-minimal python3-pycurl python3-software-properties python3.8 python3.8-minimal software-properties-common
  unattended-upgrades
Die folgenden NEUEN Pakete werden installiert:
  apache2-utils libapr1 libaprutil1 libfribidi0
0 aktualisiert, 4 neu installiert, 22 zu entfernen und 0 nicht aktualisiert.
1 nicht vollständig installiert oder entfernt.
Es müssen 488 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 427 MB Plattenplatz freigegeben.
Möchten Sie fortfahren? [Y/n]

chrisss404, 2020-03-29:
I am not sure if this package works with _Armbian_ since it is build for _Raspbian_.

However, you can check #12. Maybe **lighttpd** that is used by pi-hole is using port **80** and therefore check_mk's **apache** is not able to start. Just a guess.

Apart from that I can only advise you to use _Raspbian_ if possible.

khiro1337, 2020-03-30:
ok i try it later with armbian :) 

now i try it with raspberry - 2020-02-13-raspbian-buster-lite.img

pi@raspberrypi:~ $ sudo apt update
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.1 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0                                                                                                              MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [280 kB]
Fetched 13.3 MB in 27s (498 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
22 packages can be upgraded. Run 'apt list --upgradable' to see them.
pi@raspberrypi:~ $ sudo apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following NEW packages will be installed:
  libfl2
The following packages will be upgraded:
  binutils binutils-arm-linux-gnueabihf binutils-common bluez curl
  firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree
  firmware-realtek libbinutils libcurl4 libicu63 libpam-systemd libsystemd0
  libudev1 raspi-config rpi-eeprom rpi-eeprom-images systemd systemd-sysv udev
22 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 34.5 MB of archives.
After this operation, 2,089 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.raspberrypi.org/debian buster/main armhf bluez armhf 5.50-1                                                                                                             .2~deb10u1+rpt1 [765 kB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf firmware-atheros a                                                                                                             ll 1:20190114-1+rpt5 [3,887 kB]
Get:17 http://archive.raspberrypi.org/debian buster/main armhf firmware-brcm8021                                                                                                             1 all 1:20190114-1+rpt5 [4,360 kB]
Get:2 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf systemd-sy                                                                                                             sv armhf 241-7~deb10u3+rpi1 [99.9 kB]
Get:3 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf libpam-sys                                                                                                             temd armhf 241-7~deb10u3+rpi1 [193 kB]
Get:5 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf libsystemd                                                                                                             0 armhf 241-7~deb10u3+rpi1 [305 kB]
Get:6 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf systemd ar                                                                                                             mhf 241-7~deb10u3+rpi1 [3,251 kB]
Get:18 http://archive.raspberrypi.org/debian buster/main armhf firmware-libertas                                                                                                              all 1:20190114-1+rpt5 [3,423 kB]
Get:7 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf udev armhf                                                                                                              241-7~deb10u3+rpi1 [1,252 kB]
Get:8 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf libudev1 a                                                                                                             rmhf 241-7~deb10u3+rpi1 [143 kB]
Get:9 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf binutils-a                                                                                                             rm-linux-gnueabihf armhf 2.31.1-16+rpi2 [1,814 kB]
Get:19 http://archive.raspberrypi.org/debian buster/main armhf firmware-misc-non                                                                                                             free all 1:20190114-1+rpt5 [3,071 kB]
Get:10 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf libbinuti                                                                                                             ls armhf 2.31.1-16+rpi2 [302 kB]
Get:11 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf binutils                                                                                                              armhf 2.31.1-16+rpi2 [57.0 kB]
Get:12 http://mirror.netcologne.de/raspbian/raspbian buster/main armhf binutils-                                                                                                             common armhf 2.31.1-16+rpi2 [2,075 kB]
Get:20 http://archive.raspberrypi.org/debian buster/main armhf firmware-realtek                                                                                                              all 1:20190114-1+rpt5 [495 kB]
Get:21 http://archive.raspberrypi.org/debian buster/main armhf raspi-config all                                                                                                              20200226 [25.3 kB]
Get:22 http://archive.raspberrypi.org/debian buster/main armhf rpi-eeprom-images                                                                                                              all 5.2-1 [363 kB]
[... cut ...]
Created symlink /etc/systemd/system/multi-user.target.wants/omd.service → /etc/s                                                                                                             ystemd/system/omd.service.
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for systemd (241-7~deb10u3+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for install-info (6.5.0.dfsg.1-4+b1) ...
pi@raspberrypi:~ $

URL:
http://IP/prod

404 Not Found
The requested URL was not found on this server.

Apache/2.4.38 (Raspbian) Server at IP Port 80

khiro1337, 2020-03-30:
OK solve with raspberrry pi

pi@raspberrypi:~ $ omd status
omd: The site 'pi' does not exist. You need to execute omd as root or site user.
pi@raspberrypi:~ $ sudo omd status
pi@raspberrypi:~ $ su -
Password:
root@raspberrypi:~# omd status
root@raspberrypi:~# omd sites
SITE             VERSION          COMMENTS
root@raspberrypi:~# omd create SITENAME
Adding /opt/omd/sites/SITENAME/tmp to /etc/fstab.
Creating temporary filesystem /omd/sites/SITENAME/tmp...OK
Restarting Apache...OK
Created new site SITENAME with version 1.6.0p10.cre.

  The site can be started with omd start SITENAME.
  The default web UI is available at http://IP/SITENAME/

  The admin user for the web applications is cmkadmin with password: XXXXXXXX
  (It can be changed with 'htpasswd -m ~/etc/htpasswd cmkadmin' as site user.
)
  Please do a su - SITENAME for administration of this site.

root@raspberrypi:~# htpasswd -m ~/etc/htpasswd cmkadmin
htpasswd: cannot modify file /root/etc/htpasswd; use '-c' to create it
root@raspberrypi:~# su mon
OMD[SITENAME]:~$ htpasswd -m ~/etc/htpasswd cmkadmin
New password:
Re-type new password:
Updating password for user cmkadmin
OMD[SITENAME]:~$ exit

later one more time with Armbian :)

closed 2020-06-13 by chrisss404