chrisss404/check-mk-arm #22: 64bit releases ?

martux69, 2020-09-04:
Hi,
are there any plans for building arm64 packages (for pi4 eg) ?
Best regards
Martin

chrisss404, 2020-09-06:
Hi Martin,

at the moment I have no plans to provide 64-bit packages. Do you have any experience with the 64-bit version of Raspbian?

BR
Christian

martux69, 2020-09-06:
Hi Christian,
not real experience, I just installed a raspi 4 with ubuntu 20.04 lts and try to compile check_mk from the git sources.

chrisss404, 2020-09-07:
Very interesting, how far did you get? I guess you had to adapt some of the dependencies?

martux69, 2020-09-07:
Yes, because I'm doing a try and error it was a hard way regarding the dependencies :-)
Now I think I have it all. But in the moment I stuck at the step belonging to the themes:
`Module not found: Error: Can't resolve 'graphs' in '/mnt/checkmk/web/htdocs/js'                                                                                                                                      @ ./web/htdocs/js/mobile.js                                                                                                                                                                                                                                                                                                                                                                                                            WARNING in ./web/htdocs/js/index.js                                                                                                                                                                                 Module not found: Error: Can't resolve 'graphs' in '/mnt/checkmk/web/htdocs/js'                                                                                                                                      @ ./web/htdocs/js/index.js                                                                                                                                                                                                                                                                                                                                                                                                             WARNING in ./web/htdocs/js/index.js                                                                                                                                                                                 Module not found: Error: Can't resolve 'ntop_alerts' in '/mnt/checkmk/web/htdocs/js'                                                                                                                                 @ ./web/htdocs/js/index.js                                                                                                                                                                                                                                                                                                                                                                                                             WARNING in ./web/htdocs/js/index.js                                                                                                                                                                                 Module not found: Error: Can't resolve 'ntop_flows' in '/mnt/checkmk/web/htdocs/js'                                                                                                                                  @ ./web/htdocs/js/index.js                                                                                                                                                                                                                                                                                                                                                                                                             WARNING in ./web/htdocs/js/index.js                                                                                                                                                                                 Module not found: Error: Can't resolve 'ntop_interface_details' in '/mnt/checkmk/web/htdocs/js'                                                                                                                      @ ./web/htdocs/js/index.js                                                                                                                                                                                                                                                                                                                                                                                                             ERROR in ./web/htdocs/themes/facelift/theme.scss                                                                                                                                                                    Module build failed (from ./node_modules/sass-loader/dist/cjs.js):                                                                                                                                                  SassError: File to import not found or unreadable: ../facelift/scss/cee/_graphs.scss.                                                                                                                                       on line 37 of /mnt/checkmk/web/htdocs/themes/facelift/theme.scss      `
I think I have a wrong naming convention in the root path: I start with /mnt/checkmk. I just renamed it to /mnt/check-mk-raw-1.7.0i1.cre and try a next run.
BTW: I use the git sources ...
I will inform you about my progress when you want ?

chrisss404, 2020-09-14:
Hey, sorry for the late reply. Yes of course I am interested.

A few remarks:
* I have never built directly from the master branch before, so I have no experience in this regard.
* I would always use an official release from https://checkmk.de/download-source.php, you can never know if the master branch is as stable as is should be.
* If you've already installed all the required dependencies, you should be able to run the script from this repository like this: `bash build_check_mk.sh 1.6.0p17`

Good luck & best regards

martux69, 2020-09-16:
Hi,
during the last days I tried several things, but unfortunately without success in the end:
- compile from github
- compile from source tar
Always some dependency will not fullfilled: node-sass not available for arm64, pymssql too old, etc.
I think the main problem was that I used Ubuntu focal as OS. It seems to be too new for the tar (eg. python2 vs. python3) or some dependencies in the git.
So I will try another way and install my pi4 with an ubuntu bionic arm64 version the next days.
I will report :-)

martux69, 2020-09-20:
Hey,
I did it :-) I can compile a 64bit version of check_mk-1.6.0p17 for a raspberry pi with ubuntu 18.04 lts arm64.
The version seems working so far. I will document my steps the next days and will publish it here if it is ok for you.
Best regards

chrisss404, 2020-09-21:
Well done!
Sure, please go ahead.

ianw1974, 2020-09-25:
@martux69 I'd be interested in your steps also.  I did have Ubuntu 18.04 LTS on my RPI4 but now have 20.04.  Also noticed 1.6.0p17 is python2 and the github now has python3, so could be more successful with this for 20.04, but I'm still hitting roadblocks with it and cannot figure out how to fix the compile errors that I've been getting.  So your steps might help out so I could utilise and adapt to get 20.04 working and could also post my steps here then for how to get it working on the latest Ubuntu LTS release.

martux69, 2020-09-30:
Hi Ian,

sorry for delay in answering. I will document & publish my steps next weekend.

Best regards

Martin

martux69, 2020-10-05:
Hi Ian,
starting with the information from https://forum.checkmk.com/t/check-mk-error-in-compilation-from-source-git/16999

- download and extract  check-mk-raw-1.6.0p17.cre.tar.gz
- `make setup`
- apt install a lot of dev packages (much more than make setup did) (a list of the installed dev packages on my pi is here 
[dev-packages.log](https://github.com/chrisss404/check-mk-arm/files/5329497/dev-packages.log) )
- `apt install devscripts dpatch flex bison alien`
- remove the enterprise includes as mentioned in the link above
- `touch ./agents/windows/check_mk.user.yml`
- /usr/include/sybdb.h must be extend by this line: #define DBVERSION_80 DBVERSION_71 (see https://github.com/pymssql/pymssql/issues/520)
- in agents/Makefile remove win-packages in build section
- `touch agents/windows/check_mk_agent_arm64.{exe,msi}'
- in the following tar packages the config.guess file must be replaced by the one from /usr/share/automake-1.15/
  * omd/packages/net-snmp/net-snmp-0b32548.tgz
  * omd/packages/nagios/nagios.tgz
  * omd/packages/nsca/nsca.tgz
  * omd/packages/pnp4nagios/pnp4nagios.tgz
  * omd/packages/stunnel/stunnel.tgz
- delete navicli from omd/Makefile
- in snap7-full-1.4.2/build/unix create a copy from armv6_linux.mk as  aarch64_linux.mk and modify the content by changing armv6 to aarch64
- `make dist`
- `make deb DEBFULLNAME="<name>" DEBEMAIL=<mail>`

If all goes fine, the chech-mk deb will be created. Before installing it the following packages must be installed:
`apt install traceroute dialog graphviz apache2 apache2-utils libevent-1.4-2 libpango1.0-0 php-cli php-cgi php-gd php-sqlite3 php-json php-pear smbclient rpcbind xinetd freeradius-utils lcab libgsf-1-114 poppler-utils gdebi`
 After installing the check-mk.deb I had to create some links for activating apache2 moduls:
`ln -s /usr/lib/apache2/modules/mod_mpm_prefork.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_access_compat.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_authn_core.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_authz_core.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_filter.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_auth_basic.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_authn_file.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_authz_host.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_authz_user.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_autoindex.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_env.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_expires.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_deflate.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_headers.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_setenvif.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_mime_magic.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_mime.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_negotiation.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_dir.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_alias.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_rewrite.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_cgi.so /omd/sites/monitor/lib/apache/modules/`
`ln -s /usr/lib/apache2/modules/mod_status.so /omd/sites/monitor/lib/apache/modules/`

Now I could create the check-mk instance with omd start monitor.

I hope I got it all. It was a lot of try & error with many waiting time.
In the future I will try to do it all with a dockerfile and a cross compile in order to get the compile process faster.

I try it also on ubuntu 20.04 before, but without success. Beside the python3 problem it seems there are a lot of depending dev library being not available for aarch64 on ubuntu 20 at the moment.

Hopefully you have more success with ubuntu 20.04, I'm curious about it :-).

Kind regards
Martin

ianw1974, 2020-10-05:
@martux69 thx for the detail.  I'll try this on my RPI4 with Ubuntu 20.04 and see if I will have any success.  Will most likely try over the weekend, so will update here as soon as I can if I have success.

martux69, 2020-10-05:
@chrisss404 

if you like I can upload my aarch64 and you can published it here ?

Kind regards
Martin

chrisss404, 2020-10-06:
@martux69 sure we can do that, but I think it only makes sense if you provide packages for upcoming releases as well. Are you planning to do that?

martux69, 2020-10-07:
Hi Christian,

as I will go on using check-mk in the future I will try it, but I cannot say how fast I will be.

closed 2020-11-26 by martux69

robertoswald, 2021-01-10:
Hi, 
this is my check_mk build script working on Raspberry Pi 4 with Ubuntu 20.04 (aarch64). Maybe this is helping somebody... (it will take about 130 Minutes)
```
#!/bin/bash

VERSION="1.6.0p19"
DEB_FULLNAME="check_mk_server for Ubuntu 20.04 on RPI4"
DEB_EMAIL="your@email.foo"

# for aarch64
NETSNMP=net-snmp-0b32548
NSCA=nsca-2.9.1
NAGIOS=nagios-3.5.1
STUNNEL=stunnel-5.50
PNP4NAGIOS=pnp4nagios-0.6.26
SNAP7=snap7-full-1.4.2

### NOTHING TO CHANGE AFTER HERE

if [ $# -gt 0 ]; then
  VERSION="$1"
fi

BASEDIR=$(pwd)
HWPLATFORM=$(uname -i)

echo "-----> BUILDING CHECK_MK ${VERSION} <-----"

# install build dependencies
echo "-----> INSTALLING BUILD DEPENDENCIES <-----"
apt-get update
apt-get -y install apache2 build-essential debhelper dnsutils dpatch flex fping git \
           git-buildpackage make rpcbind rrdtool smbclient snmp apache2-dev \
	   default-libmysqlclient-dev dietlibc-dev libboost-all-dev libboost-dev \
           libdbi-dev libevent-dev libffi-dev libfreeradius-dev libgd-dev \
	   libglib2.0-dev libgnutls28-dev libgsf-1-dev libkrb5-dev libmcrypt-dev \
           libncurses-dev libpango1.0-dev libpcap-dev libperl-dev libpq-dev \
           libreadline-dev librrd-dev libsqlite3-dev libssl-dev libxml2-dev tk-dev uuid-dev

# downloading check_mk sources
if [ ! -f check-mk-raw-${VERSION}.cre.tar.gz ]; then
  echo "-----> DOWNLOADING CHECK_MK SOURCES <-----"
  wget https://mathias-kettner.de/support/${VERSION}/check-mk-raw-${VERSION}.cre.tar.gz 
fi

# unpacking check_mk sources
echo "-----> UNPACKING CHECK_MK SOURCES <-----"
if [ -d check-mk-raw-${VERSION}.cre ]; then
  rm -r check-mk-raw-${VERSION}.cre
fi
tar -xzf check-mk-raw-${VERSION}.cre.tar.gz

# modifications for aarch64
if [ $HWPLATFORM = aarch64 ]; then
  update_cfg_guess()
  {
    cp /usr/share/automake-1.16/config.guess .
    cd ..
  }

  echo "-----> UPDATE CONFIG.GUESS FOR AARCH64 <-----"
  cd $BASEDIR/check-mk-raw-${VERSION}.cre/omd/packages/net-snmp
  tar -xzf $NETSNMP.tar.gz
  cd $NETSNMP; update_cfg_guess
  rm $NETSNMP.tar.gz
  tar -czf $NETSNMP.tar.gz $NETSNMP && rm -r $NETSNMP

  cd ../nsca
  tar -xzf $NSCA.tar.gz
  cd $NSCA; update_cfg_guess
  rm $NSCA.tar.gz
  tar -czf $NSCA.tar.gz $NSCA && rm -r $NSCA

  cd ../nagios
  tar -xzf $NAGIOS.tar.gz
  cd $NAGIOS; update_cfg_guess
  cd $NAGIOS/tap; update_cfg_guess; cd ..
  rm $NAGIOS.tar.gz
  tar -czf $NAGIOS.tar.gz $NAGIOS && rm -r $NAGIOS

  cd ../stunnel
  tar -xzf $STUNNEL.tar.gz
  cd $STUNNEL; update_cfg_guess
  rm $STUNNEL.tar.gz
  tar -czf $STUNNEL.tar.gz $STUNNEL && rm -r $STUNNEL

  cd ../pnp4nagios
  tar -xzf $PNP4NAGIOS.tar.gz
  cd $PNP4NAGIOS; update_cfg_guess
  rm $PNP4NAGIOS.tar.gz
  tar -czf $PNP4NAGIOS.tar.gz $PNP4NAGIOS && rm -r $PNP4NAGIOS

  echo "-----> CREATE MISSING aarch64_linux.mk <-----"
  cd ../snap7
  tar -xzf $SNAP7.tar.gz
  cp $SNAP7/build/unix/arm_v6_linux.mk $SNAP7/build/unix/aarch64_linux.mk
  sed -i 's/arm_v6/aarch64/' $SNAP7/build/unix/aarch64_linux.mk
  rm $SNAP7.tar.gz
  tar -czf $SNAP7.tar.gz $SNAP7 && rm -r $SNAP7

  echo "-----> DISABLE NAVICLI BUILD <-----"
  sed -i 's/    navicli/\#   navicli/' $BASEDIR/check-mk-raw-${VERSION}.cre/omd/Makefile
fi

# build package
echo "-----> BUILDING PACKAGE <-----"
echo $BASEDIR
cd $BASEDIR/check-mk-raw-${VERSION}.cre
make -C omd setup
DEBFULLNAME="$DEB_FULLNAME" DEBEMAIL="$DEB_EMAIL" make deb

# cleanup
if [ $? -eq 0 ]; then
  mv check-mk-raw-${VERSION}*.deb ..
  cd ..
  rm -rf check-mk-raw-${VERSION}.cre
fi
```
Cheers
Robert

robertoswald, 2021-01-11:
Hi,
here is my build_script working on Ubuntu 20.04 on RPI4 (aarch64), maybe it is helping somebody.
It will take about 120 minutes.
```
#!/bin/bash

VERSION="1.6.0p19"
DEB_FULLNAME="check_mk_server for Ubuntu 20.04 on RPI4"
DEB_EMAIL="your@email.foo"

# for aarch64
NETSNMP=net-snmp-0b32548
NSCA=nsca-2.9.1
NAGIOS=nagios-3.5.1
STUNNEL=stunnel-5.50
PNP4NAGIOS=pnp4nagios-0.6.26
SNAP7=snap7-full-1.4.2

### NOTHING TO CHANGE AFTER HERE

if [ $# -gt 0 ]; then
  VERSION="$1"
fi

BASEDIR=$(pwd)
HWPLATFORM=$(uname -i)

echo "-----> BUILDING CHECK_MK ${VERSION} <-----"

# install build dependencies
echo "-----> INSTALLING BUILD DEPENDENCIES <-----"
apt-get update
apt-get -y install apache2 build-essential debhelper dnsutils dpatch flex fping git \
           git-buildpackage make rpcbind rrdtool smbclient snmp apache2-dev \
	   default-libmysqlclient-dev dietlibc-dev libboost-all-dev libboost-dev \
           libdbi-dev libevent-dev libffi-dev libfreeradius-dev libgd-dev \
	   libglib2.0-dev libgnutls28-dev libgsf-1-dev libkrb5-dev libmcrypt-dev \
           libncurses-dev libpango1.0-dev libpcap-dev libperl-dev libpq-dev \
           libreadline-dev librrd-dev libsqlite3-dev libssl-dev libxml2-dev tk-dev uuid-dev

# downloading check_mk sources
if [ ! -f check-mk-raw-${VERSION}.cre.tar.gz ]; then
  echo "-----> DOWNLOADING CHECK_MK SOURCES <-----"
  wget https://mathias-kettner.de/support/${VERSION}/check-mk-raw-${VERSION}.cre.tar.gz 
fi

# unpacking check_mk sources
echo "-----> UNPACKING CHECK_MK SOURCES <-----"
if [ -d check-mk-raw-${VERSION}.cre ]; then
  rm -r check-mk-raw-${VERSION}.cre
fi
tar -xzf check-mk-raw-${VERSION}.cre.tar.gz

# modifications for aarch64
if [ $HWPLATFORM = aarch64 ]; then
  update_cfg_guess()
  {
    cp /usr/share/automake-1.16/config.guess .
    cd ..
  }

  echo "-----> UPDATE CONFIG.GUESS FOR AARCH64 <-----"
  cd $BASEDIR/check-mk-raw-${VERSION}.cre/omd/packages/net-snmp
  tar -xzf $NETSNMP.tar.gz
  cd $NETSNMP; update_cfg_guess
  rm $NETSNMP.tar.gz
  tar -czf $NETSNMP.tar.gz $NETSNMP && rm -r $NETSNMP

  cd ../nsca
  tar -xzf $NSCA.tar.gz
  cd $NSCA; update_cfg_guess
  rm $NSCA.tar.gz
  tar -czf $NSCA.tar.gz $NSCA && rm -r $NSCA

  cd ../nagios
  tar -xzf $NAGIOS.tar.gz
  cd $NAGIOS; update_cfg_guess
  cd $NAGIOS/tap; update_cfg_guess; cd ..
  rm $NAGIOS.tar.gz
  tar -czf $NAGIOS.tar.gz $NAGIOS && rm -r $NAGIOS

  cd ../stunnel
  tar -xzf $STUNNEL.tar.gz
  cd $STUNNEL; update_cfg_guess
  rm $STUNNEL.tar.gz
  tar -czf $STUNNEL.tar.gz $STUNNEL && rm -r $STUNNEL

  cd ../pnp4nagios
  tar -xzf $PNP4NAGIOS.tar.gz
  cd $PNP4NAGIOS; update_cfg_guess
  rm $PNP4NAGIOS.tar.gz
  tar -czf $PNP4NAGIOS.tar.gz $PNP4NAGIOS && rm -r $PNP4NAGIOS

  echo "-----> CREATE MISSING aarch64_linux.mk <-----"
  cd ../snap7
  tar -xzf $SNAP7.tar.gz
  cp $SNAP7/build/unix/arm_v6_linux.mk $SNAP7/build/unix/aarch64_linux.mk
  sed -i 's/arm_v6/aarch64/' $SNAP7/build/unix/aarch64_linux.mk
  rm $SNAP7.tar.gz
  tar -czf $SNAP7.tar.gz $SNAP7 && rm -r $SNAP7

  echo "-----> REMOVE NAVICLI BUILD <-----"
  sed -i '/.*navicli/d' $BASEDIR/check-mk-raw-${VERSION}.cre/omd/Makefile
fi

# build package
echo "-----> BUILDING PACKAGE <-----"
cd $BASEDIR/check-mk-raw-${VERSION}.cre
make -C omd setup
DEBFULLNAME="$DEB_FULLNAME" DEBEMAIL="$DEB_EMAIL" make deb

# cleanup
if [ $? -eq 0 ]; then
  mv check-mk-raw-${VERSION}*.deb ..
  cd ..
  rm -rf check-mk-raw-${VERSION}.cre
fi
```

Cheers 
Robert

chrisss404, 2021-01-11:
Hi Robert,

wow, nice work. Well done. Would you like to create a PR so that we can merge your adaptations?

BR
Christian

robertoswald, 2021-01-11:
Hi Christian, 
sure it would be a pleasure but I need some assistance what do do, haven´t done any PRs yet.

BR
Robert

chrisss404, 2021-01-11:
Ok that's no problem. It's quite easy.

* fork this repository, click the fork button in top-right corner
* commit your changes in your forked repository
* come back to this repository and click the pull request butten. For more details, see: https://docs.github.com/en/free-pro-team@latest/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork

HTH

ianw1974, 2021-01-13:
Can confirm @robertoswald script works great!  I've ran it today on my RPI4 with Ubuntu 20.04.  This will now become even more interesting, since CheckMK 2.0.0 will be released soon (currently in beta), and is based on python3, so should be much faster as well.  Not sure how this will affect the build process, but the scripts could certainly be adapted easily enough I think.

martux69, 2021-01-13:
Hi all,

the last days I ran sucessfully a similar script from me for creating a deb of checkMK 2.0.0. beta 3 under Ubuntu 20.10 "Groovy".

If desired I can upload the script and the deb here.

Kind regards

Martin

ianw1974, 2021-01-14:
@martux69 Please do, I think it could be easily amended for us that are staying on LTS releases, so could provide help for 20.04 users.

martux69, 2021-01-17:
I uploaded my groovy deb and also my build script. It is only a hack, not beautiful ;-)