chrisss404/check-mk-arm #24: Not an issue but more of understanding

jason354007, 2020-12-25:
I used the below code(as suggested) to build the final package myself on my Ubuntu 20.04 laptop(amd64) and ran it as root:

```
$ bash build_gcc.sh
# install dependencies and build Checkmk
$ INSTALL_DEPENDENCIES=1 bash build_check_mk.sh 1.6.0.p19
```
After the build script was complete, I got the final package as check-mk-raw-1.6.0p19_0.bionic_amd64.deb instead of check-mk-raw-1.6.0p19_0.bionic_arm64.deb ? Can you please tell me where I am wrong ?

chrisss404, 2020-12-26:
Hi Jason,

this build script is not using cross-compilation and therefore it targets the host's system architecture. If you want to get an arm package you need to run it on an arm system.

For Ubuntu Bionic Beaver there are pre-built packages provided by @martux69, see [check-mk-raw-1.6.0p19_0.bionic_arm64.deb](https://github.com/chrisss404/check-mk-arm/releases/download/1.6.0p19/check-mk-raw-1.6.0p19_0.bionic_arm64.deb)

BR
Christian

jason354007, 2020-12-30:
Thanks @chrisss404  for your response. Can I also use this on my Raspberry Pi 4  which is 64 bit ?

chrisss404, 2021-01-11:
Sorry for late response. Yes this should be possible. Maybe you can try this: https://github.com/chrisss404/check-mk-arm/issues/22#issuecomment-758007603

closed 2021-02-19 by chrisss404