chrisss404/check-mk-arm #28: Check MK 2.0 on RPI 4

Rdago, 2021-03-02:
Hi there, 

just noticed that it is not possible to install 2.0.0b7 on RPI4. I´m getting multiple issues like: 

```
/bin/bash: line 4: chrpath: command not found
make[1]: *** [packages/Python3/Python3.make:47: /home/pi/check-mk-arm/check-mk-raw-2.0.0b7.cre/omd/build/stamps/Python-3.8.7-cache-pkg-process] Error 127
make[1]: Leaving directory '/home/pi/check-mk-arm/check-mk-raw-2.0.0b7.cre/omd'
make: *** [debian/rules:42: install] Error 2
dpkg-buildpackage: error: debian/rules build-arch subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage --rules-target build-arch -rfakeroot -us -uc -ui -i.git -I.git -i.gitignore -I.gitignore failed
make[1]: *** [Makefile:274: deb] Error 2
make[1]: Leaving directory '/home/pi/check-mk-arm/check-mk-raw-2.0.0b7.cre/omd'
make: *** [Makefile:98: deb] Error 2
```

Will there be a release for it soon? 

Thank you !

chrisss404, 2021-03-02:
Hi Rdago,

can you please provide some more information?

#### What distribution are you using?

For me it is Raspbian:
```
cat /etc/*-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

#### What command are you running when you try to install it?

For me it is as follows:
```
dpkg -i check-mk-raw-1.6.0p22_0.buster_armhf.deb
apt-get install -f
```

@martux69 do you have an idea what might be wrong here?

THX & BR
Christian

Rdago, 2021-03-02:
Hi, 

Version 1.6 works fine for me. I´m trying to get Version 2.0 up running.

``` 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

chrisss404, 2021-03-02:
Ah I see. I have no plans to provide pre-built packages for 2.0 for Raspian (Raspberry Pi OS) while it's in beta. However, you can try to build it yourself.

Following changes could work (no guarantee). I was able to build one of the beta versions with them:
[checkmk2_0.patch](https://github.com/chrisss404/check-mk-arm/files/6070993/checkmk2_0.patch.txt)

Rdago, 2021-03-02:
okay. 

Will you provide the 2.0 when it will be released?

chrisss404, 2021-03-02:
Yes, I plan to.

martux69, 2021-03-02:
Hi,
have you seen the 2.0 packages here on this site (marked as pre-releases) ? I build and use them successfully on a PI4 with groovy. I build also packages for focal, but I cannot test them. Feel free to try it

sectary-want-holly, 2021-03-15:
Will there be 32bit builds for 2.0?

Asepsis, 2021-03-15:
> 
> 
> Will there be 32bit builds for 2.0?

I hope so, and he said it's planed so we have to wait. You can try to build it by yourself, I faced a lot of issues, sadly most of them are appearing after two houres of compiling. I tried it the last two weeks, first with the 2.0.0beta, after with the 2.0.0 stable but no reward. I also installed qemu to simulate an arm cpu to compile faster ^^ no reward too

chrisss404, 2021-03-19:
Checkmk 2.0 for Raspberry Pi OS Buster (armhf) is released now.

closed 2021-03-19 by chrisss404