chrisss404/check-mk-arm #29: Would it be possible to build it on Raspberry Pi Model B Rev 2?

Tronde, 2021-03-07:
Hi,  have you ever tried to build check-mk-arm on one of the first Pis? Do you think that would be possible or do you know of any reason why it's not?

I read that checkmk is very efficient. But would the hardware resources of the RPi 1 B be enough to run it in a home environment?

Looking forward reading your reply.

Thanks in advance,  
Jörg

chrisss404, 2021-03-08:
Hi Jörg,

yes that's possible. I am using a RPi first gen to build the Raspberry Pi OS packages that you can find here. Give it a try, it should work even though I wouldn't recommend it.

If you decide to try it, please let us know how your RPi can handle the current version of Checkmk.

Have fun & BR
Christian

Tronde, 2021-03-08:
Hi Christian,  
Thanks for your reply. Just to be sure, I can use the package find [here](https://github.com/chrisss404/check-mk-arm/releases/tag/1.6.0p22) for the RPi first gen as well?

I'll check it out and let you know here about my experiences.

Thanks,  
Jörg

chrisss404, 2021-03-09:
Yes, if you are using Raspberry Pi OS (formerly Raspbian) "Buster" on your RPi.

Tronde, 2021-03-12:
Hi,  
I was able to install the deb-package on my RPi. But it is not possible to use it. My RPi is running constantly with a load average between 2.75 to 4.11 eating up almost all CPU cycles.

I'm going to shop a new RPi 4 and try again. ;-)

Thanks for your help,  
Joerg

closed 2021-03-12 by Tronde