chrisss404/check-mk-arm #30: No armhf architecture .deb file exists via ../check-mk-arm/releases/latest

0phoi5, 2021-03-11:
Your instructions advise to download the latest armhf architecture .deb file using the below command.
However, that file does not exist. There is only an arm64 file in that location, which does not match the architecture of the RPi3.

curl -LO $(curl -s https://api.github.com/repos/chrisss404/check-mk-arm/releases/latest | grep browser_download_url | cut -d '"' -f 4 | grep buster_armhf.deb)

0phoi5, 2021-03-11:
Your other links are now incorrect too. Where it says this...

Latest pre-built packages
Checkmk 1.6.0 for Raspberry Pi OS (32-bit) Buster: latest

... the link actually leads to the 2.0.0 version of CMK, not Buster 1.6.

0phoi5, 2021-03-11:
Managed to find the correct file via https://github.com/chrisss404/check-mk-arm/releases/tag/1.6.0p22

chrisss404, 2021-03-11:
Thanks for reporting. Updated Readme.

closed 2021-03-11 by chrisss404