chrisss404/check-mk-arm #32: Unable to parse Autocheck after Update from 1.6 to 2.0

Tibii73, 2021-03-30:
Hello,

I have the problem that for several hosts the same issue appeared and now they where UNKNOWN in Check_MK.

[image]

The Agent can be found:

[image]

When I switch to the services the same error message with unable to parse autocheck appears.

I really don´t know what causes the problem. Can anyone please help me?

Best Regards,

Tim

chrisss404, 2021-03-30:
Hi Tim,

sorry I can not help you with this issue, I have never experienced anything similar. I would advise you to ask in the [Checkmk forum](https://forum.checkmk.com), I think you could get better answers there. 

[This issue](https://forum.checkmk.com/t/not-able-to-activate-changes/18998/5) may be related, or at least it contains a similar error message. 

HTH & BR
Christian

Asepsis, 2021-03-31:
@Tibii73 I got the same error. Maybe check the String in line 6, I had an number like 10000000L, I removed the "L" and it worked.
Hope this helps you!

BR

Tibii73, 2021-03-31:
> 
> 
> @Tibii73 I got the same error. Maybe check the String in line 6, I had an number like 10000000L, I removed the "L" and it worked.
> Hope this helps you!
> 
> BR

Hello Aspesis,

thanks for your tip!
I removed the 'L' in every *.mk file and now every host has a similar error like this:

Error running automation call try-inventory (exit code 2), error:

Name 'diskstat_default_levels' is not defined in an autocheck entry of host 'HOSTNAME' (/omd/sites/sse/var/check_mk/autochecks/HOSTNAME.mk). This entry is in pre Checkmk 1.7 format and needs to be converted. This is normally done by "cmk-update-config -v" during "omd update". Please execute "cmk-update-config -v" for converting the old configuration.

I can´t run cmk-update-config -v. Everytime I try this I get the error: command not found.

BR

TIm

chrisss404, 2021-03-31:
It's strange that the update script wasn't executed for you. Did you run `omd update yoursite` when you upgraded from 1.6 to 2.0?

You can try the following to run `cmk-update-config`:
```
# su yoursite
OMD[yoursite]:~$ cmk-update-config -v
```

For your site it would be:
```
# su sse
OMD[sse]:~$ cmk-update-config -v
```

Tibii73, 2021-04-01:
> It's strange that the update script wasn't executed for you. Did you run `omd update yoursite` when you upgraded from 1.6 to 2.0?
>
> You can try the following to run `cmk-update-config`:
>
> ```
> # su [...]

Hello Chris,

thanks for you help!
I didnt know I need to log on with the user from the site. I tried it with the root user.
Now everything is working fine.
Thanks all for you help!

BR

Tim

closed 2021-04-01 by Tibii73