chrisss404/check-mk-arm #37: Does your check-mk-arm v2.0 work on an RPI4 or 3?

danmagorian, 2021-05-20:
Hello!  I'm new to Check_Mk and installed your package on 5 RPIs:  two 4s, a 3, and two 2s running the same latest Raspberry Pi OS Lite release  March 4, 2021.  All commands were done in parallel on all 5 creating mysite etc,.  Omd status shows all 5 running., but only the two RPI2s deliver the web page:  the other 3 deliver a 404 error.  Apache 2 seems to be running, and restarting it doesn't help, but it doesn't deliver the default Apache page either.  Thanks,  Dan

danmagorian, 2021-05-20:
I'm guessing there's something missing in Lite that Apache wants, but then why does it work on the two RPI2s?

chrisss404, 2021-05-21:
Hi Dan,

sorry, I have no idea. The only thing I can think of is that you can check that the omd config has been enabled:

    ls /etc/apache2/conf-enabled/zzz_omd.conf

Btw I am using a RPi3 and it is working fine for me.

BR
Christian

PIOTA89, 2021-05-29:
Hello. 

i also experienced some installation issues on RPI4. If i try to install the deb package as described, i get the following error message: dpkg: error processing package check-mk-raw-2.0.0p4 (--install):
 dependency problems - leaving unconfigured
Processing triggers for systemd (245.4-4ubuntu3.6) ...
Errors were encountered while processing:
 check-mk-raw-2.0.0p4
 
The installation was tried on a clean ubuntu 20.04.2 system on a RPI4. 

Maybe you have an idea how to fixed it? 

Thanks in advance.

chrisss404, 2021-05-31:
Hey PIOTA89,

please don't hijack unrelated issues.

Have you tried to run `apt-get install -f` after `dpkg -i check-mk-raw....deb`?

BR
Christian

PIOTA89, 2021-06-03:
Hello, 

i acutally solved the problem and check_mk runs as espected. Really awesome work. The solution is: On a RPI4 i needed to install the dependencies manually. After that, everything works fine. 

BR 
PIOTA

martux69, 2021-06-06:
The next time try to use [gdebi](https://itsfoss.com/gdebi-default-ubuntu-software-center/). It resolves dependencies automatically.

closed 2021-07-22 by chrisss404