chrisss404/check-mk-arm #38: Instructions for building from source is missing required dependencies to build

catchdave, 2021-05-21:
I had to build from source many times to discover each dependency.

The below is the summation of deps needed to build on ubuntu 21:
`sudo apt-get install binutils bison build-essential chrpath cmake debhelper devscripts flex gcc gfortran gtk-doc-tools lcab libcurl4-openssl-dev libexpat1-dev libffi-dev libglib2.0-dev libgsf-1-114 libgtk2.0-dev libgtksourceview2.0-dev libjpeg-dev libkrb5-dev libldap2-dev libncurses-dev libperl-dev libpq5 libreadline-dev libsasl2-dev libssl-dev libxml2-dev libxmlsec1-dev libyaml-cpp-dev lintian pkg-config poppler-utils python-dev python3-lasso rpm`

chrisss404, 2021-05-21:
Hi catchdave,

thanks for providing this information.

Do you mind to create a PR that adds the dependencies to the appropriate build script?

THX & BR
Christian

martux69, 2021-06-06:
I created a Dockerfile based on the one how I build the groovy and focal debs:

[Dockerfile.txt](https://github.com/chrisss404/check-mk-arm/files/6604652/Dockerfile.txt)

closed 2021-11-07 by chrisss404