chrisss404/check-mk-arm #41: Proxy Error

guif, 2021-08-12:
Hi,

After install via this on a Raspberry PI Zero:

```
curl -LO $(curl -s https://api.github.com/repos/chrisss404/check-mk-arm/releases/tags/2.0.0p8 | grep browser_download_url | cut -d '"' -f 4 | grep buster_armhf.deb) 
dpkg -i check-mk-raw-2.0.0p8_0.buster_armhf.deb
```

And check all services OK:

```
root@checkmk:~# su - monitor
OMD[monitor]:~$ omd status
mkeventd:       running
rrdcached:      running
npcd:           running
nagios:         running
apache:         running
redis:          running
crontab:        running
-----------------------
Overall state:  running
```

Appears on browser this:

```
Proxy Error
The proxy server received an invalid response from an upstream server.
The proxy server could not handle the request

Reason: Error reading from remote server

Apache/2.4.38 (Raspbian) Server at 10.0.1.16 Port 80
```

What's wrong?

chrisss404, 2021-08-15:
Hi guif,

judging from the error logs you provided in the [checkmk forum](https://forum.checkmk.com/t/error-reading-from-remote-server/26469) you are experiencing a timeout that is caused by the fact that checkmk is taking too long to respond.

You can try to increase the timeout in `/omd/sites/monitor/etc/apache/proxy-port.conf`, but most likely your RPi is not able handle this version of checkmk due to insufficient hardware resources. Issue #29 might be related.

BR
Christian

```
==> /var/log/apache2/error.log <==
[Thu Aug 12 18:36:36.900244 2021] [proxy_http:error] [pid 3978:tid 3035616288] (70007)The timeout specified has expired: [client 10.0.1.52:50800] AH01102: error reading status line from remote server 127.0.0.1:5000
[Thu Aug 12 18:36:36.949951 2021] [proxy:error] [pid 3978:tid 3035616288] [client 10.0.1.52:50800] AH00898: Error reading from remote server returned by /monitor/check_mk/
```

guif, 2021-08-21:
Thanks Christian.
I'm waiting for the new version.

closed 2022-02-17 by chrisss404