chrisss404/check-mk-arm #42: Packages for Debian Buster aarch64

thilo-adk, 2021-09-11:
Hi,
are packages planned for Debian Buster aarch64?

Regards
Thilo

chrisss404, 2021-09-13:
Hi Thilo,

at the moment I have no plans to provide 64-bit packages for Debian Buster.

You can try to use a multiarch setup, but I am not sure if it will work. See: https://wiki.debian.org/Multiarch/HOWTO

HTH & BR
Christian

thilo-adk, 2021-09-13:
> Hi Thilo,
> 
> at the moment I have no plans to provide 64-bit packages for Debian Buster.
> 
> You can try to use a multiarch setup, but I am not sure if it will work. See: https://wiki.debian.org/Multiarch/HOWTO
> 
> HTH & BR
> Christian

Hi Christian,

thanks for the Link. I am looking at this. At the moment a compilation is running on a 4 Pi ;-)

thilo-adk, 2021-09-14:
Hi,

the compilation fails with a missing permission:

`ed -e 's/###OMD_VERSION###/2.0.0p9.cre/g' \
    -e 's/###OMD_SERIAL###/38/g' \
   `pwd`/debian/postinst.in > `pwd`/debian/postinst
sed -e 's/###OMD_VERSION###/2.0.0p9.cre/g' \
   `pwd`/debian/preinst.in > `pwd`/debian/preinst
sed -e 's/###OMD_VERSION###/2.0.0p9.cre/g' \
   `pwd`/debian/prerm.in > `pwd`/debian/prerm
sed -e 's/###OMD_VERSION###/2.0.0p9.cre/g' \
   `pwd`/debian/postrm.in > `pwd`/debian/postrm
sed -e 's/###OMD_VERSION###/2.0.0p9.cre/g' \
   `pwd`/debian/init.in > `pwd`/debian/init
sed -e 's/###OMD_VERSION###/2.0.0p9.cre/g' \
   `pwd`/debian/links.in > `pwd`/debian/links
 dpkg-buildpackage --rules-target binary-arch -rfakeroot -us -uc -ui -i.git -I.git -i.gitignore -I.gitignore
dpkg-buildpackage: Information: Quellpaket check-mk-raw-2.0.0p9
dpkg-buildpackage: Information: Quellversion 0.buster
dpkg-buildpackage: Information: Quelldistribution buster
dpkg-buildpackage: Information: Quelle geändert durch Checkmk team <feedback@checkmk.com>
dpkg-buildpackage: Information: Host-Architektur arm64
 fakeroot debian/rules binary-arch
dh_testdir
dh_testroot
dh_installchangelogs
dh_installdocs
dh_installexamples
dh_installdebconf -n
dh_installinit
dh_installman
dh_link
dh_strip --exclude=wkhtmltopdf --exclude=cmc --exclude=waitmax --exclude agent_modbus
strip: unable to copy file 'debian/check-mk-raw-2.0.0p9/opt/omd/versions/2.0.0p9.cre/lib/aarch64-linux-gnu/perl5/5.28/auto/Lasso/Lasso.so'; reason: Keine Berechtigung
dh_strip: error: strip --remove-section=.comment --remove-section=.note debian/check-mk-raw-2.0.0p9/opt/omd/versions/2.0.0p9.cre/lib/aarch64-linux-gnu/perl5/5.28/auto/Lasso/Lasso.so returned exit code 1
dh_strip: error: Aborting due to earlier error
make: *** [debian/rules:103: binary-arch] Fehler 2
dpkg-buildpackage: Fehler: Unterprozess fakeroot debian/rules binary-arch lieferte Exitstatus 2
debuild: fatal error at line 1182:
dpkg-buildpackage --rules-target binary-arch -rfakeroot -us -uc -ui -i.git -I.git -i.gitignore -I.gitignore failed
make[1]: *** [Makefile:277: deb] Fehler 2
make[1]: Verzeichnis „/home/pi/Downloads/checkmk/check-mk-arm/check-mk-raw-2.0.0p9.cre/omd“ wird verlassen
make: *** [Makefile:98: deb] Fehler 2`

In the folder the permissions OK:

ll
insgesamt 1516
-r-xr-xr-x 1 pi pi 1548616 Sep 13 22:48 Lasso.so

I build with the pi user. No sudo or root.

Regards
Thilo

thilo-adk, 2021-09-14:
Strange, quote or code function does not seem to work.

dpkg-buildpackage --rules-target binary-arch -rfakeroot -us -uc -ui -i.git -I.git -i.gitignore -I.gitignore
dpkg-buildpackage: Information: Quellpaket check-mk-raw-2.0.0p9
dpkg-buildpackage: Information: Quellversion 0.buster
dpkg-buildpackage: Information: Quelldistribution buster
dpkg-buildpackage: Information: Quelle geändert durch Checkmk team <feedback@checkmk.com>
dpkg-buildpackage: Information: Host-Architektur arm64
 fakeroot debian/rules binary-arch
dh_testdir
dh_testroot
dh_installchangelogs
dh_installdocs
dh_installexamples
dh_installdebconf -n
dh_installinit
dh_installman
dh_link
dh_strip --exclude=wkhtmltopdf --exclude=cmc --exclude=waitmax --exclude agent_modbus
strip: unable to copy file 'debian/check-mk-raw-2.0.0p9/opt/omd/versions/2.0.0p9.cre/lib/aarch64-linux-gnu/perl5/5.28/auto/Lasso/Lasso.so'; reason: Keine Berechtigung
dh_strip: error: strip --remove-section=.comment --remove-section=.note debian/check-mk-raw-2.0.0p9/opt/omd/versions/2.0.0p9.cre/lib/aarch64-linux-gnu/perl5/5.28/auto/Lasso/Lasso.so returned exit code 1
dh_strip: error: Aborting due to earlier error
make: *** [debian/rules:103: binary-arch] Fehler 2
dpkg-buildpackage: Fehler: Unterprozess fakeroot debian/rules binary-arch lieferte Exitstatus 2
debuild: fatal error at line 1182:
dpkg-buildpackage --rules-target binary-arch -rfakeroot -us -uc -ui -i.git -I.git -i.gitignore -I.gitignore failed
make[1]: *** [Makefile:277: deb] Fehler 2
make[1]: Verzeichnis „/home/pi/Downloads/checkmk/check-mk-arm/check-mk-raw-2.0.0p9.cre/omd“ wird verlassen
make: *** [Makefile:98: deb] Fehler 2

chrisss404, 2021-09-14:
I think you are missing write permissions.

thilo-adk, 2021-09-15:
> I think you are missing write permissions.

Uuuups -^_^- sometimes you can't see the forest for the trees.

OK, I have corrected the rights. During the next compilation (make deb DEBFULLNAME="abc def" DEBEMAIL=abc@def.de), the permissions were set to read again and of course it broke again. How is the check-mk-raw-2.0.0p9 directory created with the files below the debian directory?

thilo-adk, 2021-09-16:
OK, I have now tested some more. The file debian/check-mk-raw-2.0.0p9/opt/omd/versions/2.0.0p9.cre/lib/aarch64-linux-gnu/perl5/5.28/auto/Lasso/Lasso.so is created during compilation and not copied as I first assumed. However, I don't understand why the permissions don't fit only for the Lasso.so file. I have looked at other library files below ..../auto that are created during compilation and for these the user has write permissions.

chrisss404, 2021-09-17:
Hi Thilo,

I am not that familiar with the internals of the checkmk build process, therefore it's hard to tell what is going on. For in-depth questions it might be better to ask directly in the checkmk forum.

What I can tell you is that I am building as root and that it works fine for me. I have checked the permissions on my build and they seem to be the same as yours except for the user.

Before the execution of the strip command:
```
ls -alt omd/debian/check-mk-raw-2.0.0p10/opt/omd/versions/2.0.0p10.cre/lib/arm-linux-gnueabihf/perl5/5.28/auto/Lasso
total 1636
drwxr-xr-x 2 root root    4096 Sep 17 07:32 .
drwxr-xr-x 3 root root    4096 Sep 17 07:32 ..
-r-xr-xr-x 1 root root 1660972 Sep 17 07:31 Lasso.so
```

After the execution of the strip command:
```
ls -alt omd/debian/check-mk-raw-2.0.0p10/opt/omd/versions/2.0.0p10.cre/lib/arm-linux-gnueabihf/perl5/5.28/auto/Lasso
total 1016
drwxr-xr-x 2 root root    4096 Sep 17 21:26 .
-rw-r--r-- 1 root root 1027428 Sep 17 21:26 Lasso.so
drwxr-xr-x 3 root root    4096 Sep 17 07:32 ..
```

Do you have any reason to avoid root for building?

HTH & BR
Christian

thilo-adk, 2021-09-18:
Hi Christian,

> Do you have any reason to avoid root for building?

I learned this way years ago when I started with Linux. You only do the most necessary as root and that is only make install when compiling. Also packaging according to "Debian" uses fakeroot. 

I think it is still too early for Debian Buster arm64. I have other things that do not work as they should. Debian Buster arm64 still has beta status ;-) I'm thinking about switching to Ubuntu 20.04 arm64. Even if USB boot doesn't work ootb with the Pi 4.

Regards
Thilo

chrisss404, 2021-09-19:
> I learned this way years ago when I started with Linux. You only do the most necessary as root and that is only make install when compiling. Also packaging according to "Debian" uses fakeroot.

I agree, it's always better to perform tasks with as few privileges as possible. However, you can try to make it work first and then you can reduce them.

> I think it is still too early for Debian Buster arm64. I have other things that do not work as they should. Debian Buster arm64 still has beta status ;-) I'm thinking about switching to Ubuntu 20.04 arm64. Even if USB boot doesn't work ootb with the Pi 4.

There is an unmaintained build script that worked for checkmk 2.0.0b6, see: https://github.com/chrisss404/check-mk-arm/blob/1.6.0/build_check_mk_buster64.sh

Maybe this helps?

Good luck & have a nice day.

closed 2022-02-17 by chrisss404