chrisss404/check-mk-arm #46: Installation not possible on latest Raspberry OS based on Bullseye

magnum61, 2021-11-12:
Tried to install CheckMK today on a fresh installation.
Raspberry OS/Raspbian is now based on Debian 11

On this, the installation is obviously not possible.
The install reported unmet dependencies
Running apt install -f does not resolve the dependencies, it only removes the checkmk package

Any suggestions?

sergiodadi, 2021-11-12:
Hello,

Le ven. 12 nov. 2021 à 09:01, Uwe ***@***.***> a écrit :

> Tried to install CheckMK today on a fresh installation.
> Raspberry OS/Raspbian is now based on Debian 11
>
> On this, the installation is obviously not possible.
> The install reported unmet dependencies [...]

Wait for check-mk-raw-2.0.0p13_0.bullseye_armhf.deb ? 😉
-- 
Serge

chrisss404, 2021-11-12:
Hi Uwe,

you can compile a package for your system as suggested in the readme, or you can wait for a package that supports your system. However, at the moment I can't tell when this will happen.

BR
Christian

closed 2021-11-12 by chrisss404

chrisss404, 2021-11-12:
In case you need it, the link to the Raspberry OS Buster image: https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-full.zip