chrisss404/check-mk-arm #49: Package 2.0.0p21 for buster 32 bits

sergiodadi, 2022-03-08:
Hello,

using **chekmk** professionally, I am now retired and now use checkmk, on a **Raspberry Pi 4**, for my small personal network, thanks for your packages. If I understand correctly, you no longer build packages for 32-bit buster ? 

So I tried to build the packages through your scripts. For information, these packages are available [ici], (https://github.com/sergiodadi/check-mk-raspberry), this may be of interest to other people.

I upgraded from checkmk *2.0.0p17* to *2.0.0p21* on 2 differents **Raspberry** and everything was OK.

By the way, do you plan to build packages for **64-bits bulleyes**?

Thanks again for your work

Serge

herzkerl, 2022-03-09:
Thanks to the both of you—I struggled installing the newest build von 64-bit (lite) Bullseye... That explains it.
@chrisss404 Since 64-bit is the new standard in 'Raspberry Pi Imager'—mind putting a hint in the README.md? :)

sergiodadi, 2022-03-09:
Christian,

> Let me know if you want me to link your packages in this repository's README

Yes, of course, you can put a link to my GitHub. I can build the futures buster 32 bits versions .

thanks

Serge

sergiodadi, 2022-03-09:
Hi Ben,

> Thanks to the both of you—I struggled installing the newest build von 64-bit (lite) Bullseye... 

I have quickly tried  build a 64 bits Bulleyes version with Christian  script, but was unsuccessful. I tried with python-make-set-aarch64-architecture.patch.

Did you succeed ? Did you modify the patch file?

Serge

martux69, 2022-03-09:
Hi,
I built a 64-bit version for bullseye, you can find it under the release. It is untested! Feel free to test ;-)
Regards
Martin

sergiodadi, 2022-03-09:
> Hi,
> I built a 64-bit version for bullseye, you can find it under the release. It is untested! Feel free to test ;-)
> Regards
> Martin

Thanks, I quickly install on a *Bulleye 64 bits* test VM (Proxmox on Raspberry) and it 's seem OK 👍

**lsb_release -a**
- No LSB modules are available.
- Distributor ID:	Debian
- Description:	Debian GNU/Linux 11 (bullseye)
- Release:	11
- Codename:	bullseye

**uname -a**
Linux Host-003 5.10.0-11-arm64 #1 SMP Debian 5.10.92-2 (2022-02-28) aarch64 GNU/Linux

**omd status** 
Doing 'status' on site monit64:
mkeventd:       running
rrdcached:      running
npcd:           running
nagios:         running
apache:         running
redis:          running
crontab:        running
Overall state:  running

**omd versions**
2.0.0p21.cre (default)

Best regard

Serge

chrisss404, 2022-03-09:
@herzkerl 
> mind putting a hint in the README.md? :)

I updated the README.md, better? 

@sergiodadi 

> Yes, of course, you can put a link to my GitHub. I can build the futures buster 32 bits versions .

I added a link to your repository.

> Did you succeed ? Did you modify the patch file?

If you are still interested in the 64-bit build, this unmaintained build script might help. It worked for checkmk 2.0.0b6, see: https://github.com/chrisss404/check-mk-arm/blob/1.6.0/build_check_mk_buster64.sh

@martux69 

> I built a 64-bit version for bullseye

Nice!

herzkerl, 2022-03-10:
@chrisss404 

> I updated the README.md, better?

Perfect! :)

sergiodadi, 2022-03-10:
Christian,

> If you are still interested in the 64-bit build, this unmaintained build script might help. It worked for checkmk 2.0.0b6,
>  see: https://github.com/chrisss404/check-mk-arm/blob/1.6.0/build_check_mk_buster64.sh

Hello, thank you for this script that I did not know. A quick test shows that it no longer works, I will try to understand why . I think there is a problem with architecture (x86_64) somewhere...

[...]
/usr/local/src/check-mk-raw-2.0.0p21.cre/omd/build/intermediate_install/Python-3.8.7/lib/python3.8/lib-dynload/zlib.cpython-38-aarch64-linux-gnu.so: new RUNPATH: /omd/versions/2.0.0p21.cre/lib

sed -i 's|--rpath,/omd/versions/[^/]*/lib|--rpath,/omd/versions/2.0.0p21.cre/lib|g' \
    /usr/local/src/check-mk-raw-2.0.0p21.cre/omd/build/intermediate_install/Python-3.8.7/lib/python3.8/_sysconfigdata__linux_**x86_64-linux**-gnu.py

sed: can't read /usr/local/src/check-mk-raw-2.0.0p21.cre/omd/build/intermediate_install/Python-3.8.7/lib/python3.8/_sysconfigdata__linux_x86_64-linux-gnu.py: **No such file or directory**

make[1]: *** [packages/Python3/Python3.make:50: /usr/local/src/check-mk-raw-2.0.0p21.cre/omd/build/stamps/Python-3.8.7-cache-pkg-process] Error 2

make[1]: Leaving directory '/usr/local/src/check-mk-raw-2.0.0p21.cre/omd'
make: *** [debian/rules:42: install] Error 2
dpkg-buildpackage: error: debian/rules build-arch subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage --rules-target build-arch -rfakeroot -us -uc -ui -i.git -I.git -i.gitignore -I.gitignore failed
make[1]: *** [Makefile:284: deb] Error 2
make[1]: Leaving directory '/usr/local/src/check-mk-raw-2.0.0p21.cre/omd'
make: *** [Makefile:97: deb] Error 2
[...]

thank you

Serge

closed 2022-03-24 by chrisss404