chrisss404/check-mk-arm #50: Can't install .deb on Ubuntu 22.04 VM, MacBook M1

sticsk, 2022-05-27:
When doing `dpkg -i` on the deb file, i get the dependency error. 

This is the error when I execute `apt-get install -f`

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Correcting dependencies... Done
The following packages will be REMOVED:
  check-mk-raw-2.0.0p24
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 541 MB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 183744 files and directories currently installed.)
Removing check-mk-raw-2.0.0p24 (0.impish) ...
needrestart is being skipped since dpkg has failed```

martux69, 2022-05-27:
Til now there is no deb version available for Ubuntu 22.04 and it is not possible to use the version from 20.04.

sticsk, 2022-05-27:
I understand then, I tought's working anyways. Thank you

closed 2022-05-27 by sticsk

martux69, 2022-05-27:
@sticsk : I added the actual version 2.0.0.p25 of checkmk for Ubuntu 22.04, which was released yesterday. Til now it is untested, feel free to try it.