chrisss404/check-mk-arm #52: cmk-agent-ctl on arm?

obazda20, 2022-07-15:
Wow I appreciate your work very much!
I use CheckMK in the X64 variant, but I would like to install the agent on my Raspberry's as systemd deamon.

So i extracted from their packages the linux client cmk-agent-ctl and unfortunately it does not work under aarch64 (rpi4). Is it possible to build this also for arm64 or armhf?

chrisss404, 2022-07-15:
Hi obazda20,

`cmk-agent-ctl` is part of the checkmk agent package. This package is supposed to be architecture independent, see the package's control file (**Architecture: all**):

```
Package: check-mk-agent
Version: 2.1.0p6-1
Architecture: all
Maintainer: feedback@checkmk.com
Installed-Size: 4424
Section: alien
Priority: extra
Description: Checkmk Agent for Linux
  The Checkmk Agent for Linux provides information about the system.
  This can be used to monitor the host via Checkmk.
```

Therefore, it seems to be a bug that is unrelated to this project and I would like to redirect you to https://checkmk.com/community/get-started

HTH & BR
Christian

obazda20, 2022-07-15:
Hi Christian,
you are right, but a part of check-mk-agent package is cmk-agent-ctl (/usr/bin/cmk-agent-ctl). The main Script(bash) /usr/bin/check_mk_agent is working on arm. The new cmk-agent-ctl is necessary for systemd and TLS registriation and this binary does not work on arm. The workarounds are use SSH or Xinited to monitor rpi's. But my question is, is it possible to compile it on arm and run. 

It looks like the company behind Checkmk does not work on it:
https://forum.checkmk.com/t/2-1-0b1-cmk-agent-ctl-not-working-on-arm-only-on-x86-64/30062

chrisss404, 2022-07-15:
> is it possible to compile it on arm and run.

I tried to set `CARGO_BUILD_TARGET` to `arm-unknown-linux-gnueabihf` in `agents/cmk-agent-ctl/cargo_env`, however without success. Maybe someone else has another idea / hint?

I still believe that this not the right place to address this.

martux69, 2022-10-11:
Regarding to the checkmk documentation the new checkmk agent is only available for checkmk hosts running under x86_64:

```
Requirements for the host

Registering with the Agent Controller requires a Linux system with an init system systemd version 220 or later and an x86_64 computer architecture. See the Testing Agent Controller and system environment section to learn how to verify these prerequisites.
```
https://docs.checkmk.com/latest/en/agent_linux.html#_requirements_for_the_host

closed 2022-10-17 by chrisss404