chrisss404/check-mk-arm #54: armv7 - ubuntu:jammy

JzJad, 2022-12-30:
Was wondering if you could possibly add a armv7 package for ubuntu:jammy. Not required or anything it would just be nice to have so that I can package multi-arch together as I'm currently building 2 images for armv7/armv8 (bullseye/ubuntu)

Thanks for the work you have already done! 😄

chrisss404, 2023-01-03:
Hi Jimmy,

currently I have no plans to add more packages. 

However, it should be fairly easy to compile the package yourself in your docker container.  You only have to add the `omdlib-reduce-certificate-maximum-validity-period` patch and replace the `python-make-set-aarch64-architecture` with the `python-make-set-arm-architecture` patch in `build_check_mk_ubuntu_64bit.sh`.

#### Note to myself:

To build _arm64_ images on _amd64_, ensure that `qemu-user-static` is available, then:
```
# Dockerfile
FROM ubuntu:jammy

RUN dpkg --print-architecture # should print arm64
```
```
$ docker buildx build --platform linux/arm64 --progress plain --no-cache -t checkmk .
```

Good luck & BR
Christian

JzJad, 2023-01-06:
Awesome thanks for the quick tips that should make for a quick an easy solution for me to put together!

Kind Regards,
Jimmy

closed 2023-01-06 by JzJad