chrisss404/check-mk-arm #55: Oracle cmk-agent-ctl

lorivaro, 2023-04-12:
Hi, I have followed thes steps described in #53 on my Oracle Cloud instance using 2.1.0p24 (2.1.0p25 wasn't built somehow), as I'd like to monitor this server from elsewhere. 
However, I can't seem to use the `cmk-agent-ctl` command to register this server. 
Is there a way I could using this build, or am I in the wrong place?

```
louk@oracle:~$ uname -a
Linux oracleLouk 5.15.0-1032-oracle #38-Ubuntu SMP Thu Mar 23 19:30:10 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

louk@oracle:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy

louk@oracle:~$ cmk-agent-ctl
cmk-agent-ctl: command not found
```

MarsellusW, 2023-04-12:
The package is the checkmk server package, not the agent...

lorivaro, 2023-04-14:
Cheers for the clarification. I seemed to have missed the part that explains the legacy mode.