chrisss404/check-mk-arm #57: Update-Fehler 2.2.0p2 zu 2.2.0p3

AlienLifeForm1, 2023-06-16:
Ich habe folgenden Fehler beim Update:

2023-06-16 01:10:47 - Updating site 'rpi' from version 2.2.0p2.cre to 2.2.0p3.cre...

Creating temporary filesystem /omd/sites/rpi/tmp...OK
Executing update-pre-hooks script "01_mkp-disable-outdated"...OK
Executing update-pre-hooks script "02_cmk-update-config"...
-| Traceback (most recent call last):
-|   File "/omd/sites/rpi/bin/cmk-update-config", line 8, in <module>
-|     from cmk.update_config.main import main
-|   File "/omd/sites/rpi/lib/python3/cmk/update_config/main.py", line 41, in <module>
-|     from cmk.gui.wsgi.blueprints.global_vars import set_global_vars
-|   File "/omd/sites/rpi/lib/python3/cmk/gui/wsgi/blueprints/global_vars.py", line 21, in <module>
-|     from cmk.gui.wsgi.applications.checkmk import get_mime_type_from_output_format, get_output_format
-|   File "/omd/sites/rpi/lib/python3/cmk/gui/wsgi/applications/__init__.py", line 6, in <module>
-|     from cmk.gui.wsgi.applications.checkmk import CheckmkApp
-|   File "/omd/sites/rpi/lib/python3/cmk/gui/wsgi/applications/checkmk.py", line 40, in <module>
-|     from cmk.gui.wsgi.applications.utils import (
-|   File "/omd/sites/rpi/lib/python3/cmk/gui/wsgi/applications/utils.py", line 20, in <module>
-|     from cmk.gui import login, pages, userdb
-|   File "/omd/sites/rpi/lib/python3/cmk/gui/login.py", line 32, in <module>
-|     from cmk.gui.main import get_page_heading
-|   File "/omd/sites/rpi/lib/python3/cmk/gui/main.py", line 16, in <module>
-|     from cmk.gui.sidebar import SidebarRenderer
-|   File "/omd/sites/rpi/lib/python3/cmk/gui/sidebar/__init__.py", line 63, in <module>
-|     from cmk.gui.werks import may_acknowledge
-|   File "/omd/sites/rpi/lib/python3/cmk/gui/werks.py", line 19, in <module>
-|     from cmk.utils import werks as utils_werks
-|   File "/omd/sites/rpi/lib/python3/cmk/utils/werks/__init__.py", line 32, in <module>
-|     from .werkv2 import load_werk_v2, RawWerkV2
-|   File "/omd/sites/rpi/lib/python3/cmk/utils/werks/werkv2.py", line 10, in <module>
-|     import markdown
-| ModuleNotFoundError: No module named 'markdown'
ERROR (exit code: 1)

Wieso fehlt das Python Modul markdown?

Rufe ich Check_MK auf, bekomme ich erwartungsgemäß einen Fehler:

Internal Server Error
The server encountered an internal error or misconfiguration and was unable to complete your request.

Please contact the server administrator at nobody@localhost to inform them of the time this error occurred, and the actions you performed just before this error.

More information about this error may be available in the server error log.

Ich habe mir so einiges an Logs angesehen, aber noch nicht das richtige gefunden woraus hervorgeht, was jetzt weiterhin sein Problem ist.

Ein erneutes Updaten der Site funktioniert nicht bzw. läuft unendlich. Wie bekomme ich die Site wieder ans laufen? Sollte ich meinen Pi auf die letzte Vollsicherung mit Check_MK 2.1.0p29 zurücksetzen? Oder evtl. ein omd --force init?

Danke für jede Hilfe.

chrisss404, 2023-06-16:
Hi there,

> Danke für jede Hilfe.

Rerunning `omd update` should downgrade your site to the previous version and you should be able to use it again, e.g.:

```
Updating site 'health' from version 2.2.0p3.cre to 2.2.0p2.cre...
```

> Wieso fehlt das Python Modul markdown?

I have no idea, maybe it's better to skip 2.2.0p3 and wait for 2.2.0p4.

HTH & BR

chrisss404, 2023-06-17:
I did some digging, results as follows:

* `werkv2.py` was newly introduced with **2.2.0p3** and it seems that it has introduced a regression
* removing the markdown imports resolves the issue (for me):
```
vim /omd/versions/2.2.0p3.cre/lib/python3/cmk/utils/werks/werkv2.py
-import markdown
-from markdown.extensions import Extension
-from markdown.treeprocessors import Treeprocessor
```

Further notes:

* as far as I know there are only **arm64** packages of **2.1.0p29** and there is only an **armhf** package of **2.2.0p3** (for now), I suggest to stick with one architecutre and not to mix them
* checkmk 2.2.0 will be the last major version that supports armhf, so please consider switching to arm64
* I am currently low on time -> less test resources on my side, sorry for any inconveniences
* contributions are always welcome

commit fe963678 (2023-06-17): fixes #57
closed 2023-06-17 by chrisss404