chrisss404/check-mk-arm #60: in 2.2.0p6_0.bullseye_arm64 version the windows agent is an empty file. Filesize 0

jkehrel, 2023-07-21:
Maybe others are empty too.
check_mk_agent.msi

chrisss404, 2023-07-21:
Hi there,

the windows artifacts are skipped on purpose as they cannot be built on a linux host, see https://github.com/chrisss404/check-mk-arm/blob/master/build_check_mk_debian_32bit.sh#L60

Please get the windows agent from another source, e.g.:
```
$ wget https://download.checkmk.com/checkmk/2.2.0p6/check-mk-cloud-2.2.0p6_0.kinetic_amd64.deb
$ ar x check-mk-cloud-2.2.0p6_0.kinetic_amd64.deb
$ tar -I zstd -xf data.tar.zst 
$ ls opt/omd/versions/2.2.0p6.cce/share/check_mk/agents/windows/
baked_container.msi  check_mk_agent.msi  CONTENTS  OpenHardwareMonitorCLI.exe  plugins         python-3.cab
cfg_examples         check_mk.user.yml   mrpe      OpenHardwareMonitorLib.dll  python-3.4.cab  unsign-msi.patch
```

HTH & BR
Christian

jkehrel, 2023-07-21:
Thanks for your notice. I got them.
I only opened an issue on this because placeholders with file size 0 are still there. If you go back to the check-mk-raw-2.1.0p29_0.bullseye_arm64.deb file, then the real Windows files are inside. Why did you change to Zero-Files in the 2.2.0 release?
Greetings
Jürgen

chrisss404, 2023-07-23:
> I got them.
> I only opened an issue on this because placeholders with file size 0 are still there.

Ok, great. Thx for reporting.

> Why did you change to Zero-Files in the 2.2.0 release?

No special reason. It's caused due to an upstream change in the build process that requires additional steps to obtain the windows agent artifacts. However, I agree that this could be improved.

martux69, 2023-09-08:
Starting with 2.2.0p9 also the arm64 versions contains the windows agent

closed 2023-09-08 by martux69