chrisss404/check-mk-arm #61: provide version for debian bookworm

yennor, 2023-08-12:
Would there be chance you could also provide a version for debian bookworm?
The debian bullseye version doesn't properly run under bookworm, because of some old dependencies.

chrisss404, 2023-08-13:
Hi yennor,

> Would there be chance you could also provide a version for debian bookworm?

I can only speak for the 32-bit (armhf) package, there are no plans to support Debian Bookworm anytime soon.

BR
Christian

martux69, 2023-08-14:
I will try to build a 64bit version for bookworm soon.

yennor, 2023-08-15:
the 64bit version is what I was hoping for :-)
thanks

martux69, 2023-08-16:
64bit version for bookworm is added (but untested :-) )

yennor, 2023-08-16:
that was quick. Thanks.
I'll test it, but probably will take 1 to 2 weeks until I have time to do so.