chrisss404/check-mk-arm #62: Dependency resolution issue with 2.2.0p11

chrisss404, 2023-10-11:
Hi all,

recently I am running into dependency resolution issues when trying to build checkmk 2.2.0p11:
```
Locking Failed!
  Locking...
ERROR:pip.subprocessor:[present-rich] Getting requirements to build wheel exited with 1
[ResolutionFailure]:   File "/usr/local/lib/python3.11/site-packages/pipenv/resolver.py", line 811, in _main
[ResolutionFailure]:       resolve_packages(
...
[pipenv.exceptions.ResolutionFailure]: Warning: Your dependencies could not be resolved.
```
See full build log: [build.log](https://github.com/chrisss404/check-mk-arm/files/12873429/build.log)

I have no idea what is causing this, any hint is welcome.

Thx in advance & BR

martux69, 2023-12-17:
That's a pity :-( If I understand it aright from the build log, the failure is caused by compiling openssl-1.1.1u with bazel. Perhaps it could be a solution to use the openssl from the repos of the raspi os?

chrisss404, 2024-01-15:
Thx, for taking a look.

> Perhaps it could be a solution to use the openssl from the repos of the raspi os?

I am not sure how to do that.

> Perhaps also the bazel version could be a problem (it is very old)?

Yes, the bazel version is very old because bazel also has no official 32bit arm support, see subsequent project: https://github.com/koenvervloesem/bazel-on-arm

Maybe it's time to move to 64-bit and leave 32-bit behind.

closed 2024-01-15 by chrisss404