chrisss404/check-mk-arm #64: Redis seg fault on Raspberry Pi 5

solarisfire, 2024-01-31:
Running Pi OS (based on Debian Bookworm) - I know it's not directly supported, just wondering if you had any ideas?

<pre>
Starting agent-receiver...OK
Starting mkeventd...OK
Starting rrdcached...OK
Starting npcd...OK
Starting nagios...OK
Starting apache...OK
Starting redis...<jemalloc>: Unsupported system page size
<jemalloc>: Unsupported system page size
/omd/sites/pifive/etc/rc.d/85-redis: line 10: 336019 Segmentation fault      $DAEMON $OPTS
OK
Initializing Crontab...OK
</pre>

solarisfire, 2024-01-31:
Hmmm, I installed redis-server and copied the binary from /bin/redis-server into /omd/sites/pifive/bin and it starts up just fine now... Seems to be an issue with the compiled redis-server binary...?

martux69, 2024-01-31:
I think, it relates to this : https://github.com/jemalloc/jemalloc/issues/467#issuecomment-1876990263
I have no pi5 so I'm not able to test anything relating to this.

realiter, 2024-01-31:
there is a link to: https://bugzilla.redhat.com/show_bug.cgi?id=2240293 
"[Bug 2240293] Redis fails to start on aarch64 with 16kb pages" where a patch is mentioned.

martux69, 2024-02-05:
Yes, it seems so be the same solution in both urls. But as mentioned before I have no pi5 and as far as I can see with the implemented patch I could have trouble to run with a pi4.

chrisss404, 2024-03-23:
Won't fix.

closed 2024-03-23 by chrisss404