chrisss404/check-mk-arm #65: Build instructions

FloTheSysadmin, 2024-05-26:
Hi, I know the Project is discontinued, but I would like to try to build it with a new Version. Could you provide me with instruction to build it from the Source code in the repo? Help would be greatly appreciated. Thanks

chrisss404, 2024-05-26:
> Hi

Hey Flo,

> I would like to try to build it with a new Version

What have you tried so far? What are you struggling with?

> Could you provide me with instruction to build it from the Source code in the repo?

Sure. Basically you just have to follow the steps from here: https://github.com/Checkmk/checkmk?tab=readme-ov-file#building-on-your-own-packages

Whenever you run into a platform specific problem, like navicli has no arm support, then you need to find a workaround e.g. exclude the module from the build as it is done with following patch: https://github.com/chrisss404/check-mk-arm?tab=readme-ov-file#remove-module-navicli

You can find all required patches for checkmk 2.2.0 in this repo, however, there might have been changes in the newest version which are unaddressed. Meaning you have to figure out how to get around them yourself.

Good luck & BR
Christian

FloTheSysadmin, 2024-05-29:
Thank you a lot

FloTheSysadmin, 2024-06-01:
I've been trying for like three days now, but I cant do it. I have tried tu build the latest checkMK version "clean" (without the patches) and  it still fails. I have tried three ubuntu version in a vm but It always fails at 

"/root/.cache/bazel/_bazel_root/8f662310c8318904e664eac3c24d5987/external/redis/BUILD.bazel:15:8 Executing genrule @@redis//:build failed: (Exit 1): linux-sandbox failed: error executing Genrule command" 

If you use docker could you be so kind and provide me with the Dockerfile or the build Instructions that you used , since there seems to be almost no documentation on building checkMK from source.

Thank you

~Flo

chrisss404, 2024-06-01:
> If you use docker could you be so kind and provide me with the Dockerfile or the build Instructions that you used , since there seems to be almost no documentation on building checkMK from source.

I didn't, I always built it directly on the Pi using the bash script you can find in this repository.

However, Martin (@martux69) did, you can find his setup for Checkmk 2.2.0 here: https://github.com/chrisss404/check-mk-arm/tree/2.2.0 / [Dockerfile](https://github.com/chrisss404/check-mk-arm/blob/2.2.0/Dockerfile). I don't know its current state, but maybe there is something in there that can help you.

FloTheSysadmin, 2024-06-01:
Thank you for your help. I will try this.

martux69, 2024-06-16:
I added my last working dockerfiles for building checkmk-2.2 for bookworm and jammy 64bit and also the patches for it. The patches zip also include the build script I used for building the deb.

FloTheSysadmin, 2024-06-19:
> I added my last working dockerfiles for building checkmk-2.2 for bookworm and jammy 64bit and also the patches for it. The patches zip also include the build script I used for building the deb.

Thank you

martux69, 2024-07-01:
@FloTheSysadmin After `docker build` I ran it with this command (eg. for bookworm):
`podman run -ti -v /opt/build-checkmk/debs:/opt/build-mk/debs -v /opt/build-checkmk/bookworm-2.2/root:/root -v /opt/build-checkmk/patches:/opt/build-mk/patches 3bc7d266b7a6`

yennor, 2024-07-08:
@FloTheSysadmin If you manage to build it, could you tell here so? and maybe write down the steps you've taken?
I've utterly failed myself a few weeks ago...

FloTheSysadmin, 2024-07-08:
> @FloTheSysadmin If you manage to build it, could you tell here so? and maybe write down the steps you've taken?
> I've utterly failed myself a few weeks ago...

Hi, I have forked the project to work on tinkering with the patches. I will document the build instructions. If you need the latest (as of now) 2.2.0 version for aarch64 look into the releases of my fork.

yennor, 2024-07-11:
@FloTheSysadmin 
Thanks a lot. worked like a charm. I used this as an opportunity to update to bookworm ;-).

FloTheSysadmin, 2024-07-13:
> @FloTheSysadmin 
> Thanks a lot. worked like a charm. I used this as an opportunity to update to bookworm ;-).

Hi, glad it was of use to you :)

FloTheSysadmin, 2024-09-09:
Hi @chrisss404 would you be so kind and maybe link to my fork of the project somewhere in the Description? So that  Users can upgrade to the latest 2.2.0 version, since it contains important security fixes. Thank you.

chrisss404, 2024-09-09:
@FloTheSysadmin sure, would you mind creating a PR with the change?

FloTheSysadmin, 2024-09-11:
@chrisss404 Sure, but It would just be a one-liner that says that I maintain it "best effort".

chrisss404, 2024-09-15:
Closing, as the question about the build instructions should be answered.

closed 2024-09-15 by chrisss404