chrisss404/check-mk-arm #66: Retirement Announcement in README is not correct regarding Checkmk and ARM support

robin-checkmk, 2024-07-24:
Hi @chrisss404 and sorry for the direct ping.

I just stumbled across your announcement in the README, that you retire this project.
It is sad to read something like that, but I do not question your decision. Quite contrary, I am more than aware, that sometimes time is the most limiting factor on these kinds of project. So first and foremost: All the best to you on your journey and thank you so much keeping this up for so long!

There is just one thing, that is problematic: We never announced support for the Raspberry 5 or any Raspberry for that matter. I am not up-to-date, what exact CPUs are used by current Raspberries, I just want to make sure, your users know what to expect. And [our blog post](https://checkmk.com/blog/checkmk-our-road-arm-support) had a pretty good **TL;DR** on that:

> # TL;DR
> 
>    - Checkmk recognizes the potential of ARM as well as the needs of our customers
>    - Our focus is on seamless monitoring, so agents for ARM will be the starting point
>    - Operating the Checkmk server
>       - In the development of Checkmk, we continue to work towards platform neutrality and will thus achieve ARM compatibility over time. However, we are not currently focused on this task and can therefore not give a concrete time frame
>       - A focus on modern server processors: For reasons of optimal performance, no support for single board computers (such as the Raspberry Pi) can be envisaged

Let me break this down even more:
- Agents would be the first to receive ARM support
- Servers would be way down the road
- There is no timeline for this in the first place, so it could be years out

So I guess what I am asking is this: Please clarify your announcement, so users do not get the impression, that they might get Checkmk servers on ARM as soon as next year. If support for Checkmk servers on ARM comes at all, it will be years in the future.

chrisss404, 2024-07-25:
Hi Robin,

I don't think there is anything wrong with my statement, however, I am open to suggestions. Would you mind creating a PR with the corrections?

For reference:

> More than twice as fast and infinitely smoother
> Raspberry Pi 5 features the Broadcom BCM2712 quad-core Arm Cortex A76 processor @ 2.4GHz

https://www.raspberrypi.com/products/raspberry-pi-5/

> Cortex-A76 Specifications
> Architecture: Armv8-A (Harvard)
> Extensions: Armv8.1 extensions, Armv8.2 extensions

https://developer.arm.com/Processors/Cortex-A76

> Not all ARMs are created equal
>
> If we are already porting to a new processor family, we see no reason to ensure backward compatibility within this. There are indications that the ARMv8.2-A instruction set has good market penetration on the one hand, and on the other hand has features that will benefit the performance of Checkmk.
> 
> As a comparison: Cortex A72 cores are very popular for single board computers (SBCs) such as the Raspberry Pi. These still use the very first version of ARM's 64-bit instruction set, ARMv8-A, introduced in 2012. A performance penalty on current professional hardware in order to be compatible with SBCs is simply not justifiable for customers with large Checkmk installations.

https://checkmk.com/blog/checkmk-our-road-arm-support

Thx & BR
Christian

robin-checkmk, 2024-07-26:
My main points are:

- We never announced explicit support for RPi 5. It might work, but we will not guarantee it.
- There is no timeline attached to this, so the AMR builds might still be years out.

So while I appreciate your positive way of reading the blog post, I do not want to create false hopes. But it is your project after all, I just wanted to avoid misunderstandings. I filed #67 with an idea of how to phrase it.

chrisss404, 2024-07-26:
> We never announced explicit support for RPi 5.

Sure, but it would be surprising if it wouldn't work, since the RPi 5 supports the instruction set you are aiming for.

> There is no timeline attached to this, so the AMR builds might still be years out.

Seems like business is not fully committed to its announcement, it would surprise me if such a project would take longer than 6 months.

> I filed #67 with an idea of how to phrase it.

Thanks, merged.

closed 2024-07-26 by chrisss404

robin-checkmk, 2024-07-29:
> Sure, but it would be surprising if it wouldn't work, since the RPi 5 supports the instruction set you are aiming for.

I agree, but having worked for a software company for time now, I know how important precise communication is. Just a tiny misunderstanding can create a lot of frustration.

> Seems like business is not fully committed to its announcement, it would surprise me if such a project would take longer than 6 months.

We are, but both the exact details are not set in stone yet and the starting point of the actual development work. Hence, my remark.

Thanks for being so open about this!